raspberrypi/pico-sdk #2176: Pico W Occasional Crash in cyw43_arch_deinit

gazzyt, 2025-01-11:
I have an application using SDK 2.1.0 that does the following:
* reads some sensors
* joins the wifi network
* publish the readings to mqtt
* disconnect from the wifi network
* sleep for 10 mins
* repeat for ever

Source code is here: https://github.com/gazzyt/pico-weather-mqtt

This is using `pico_cyw43_arch_lwip_threadsafe_background mode` of the networking stack.

The application runs fine mostly but I had the following issue after about 150 cycles.
I can't see anything in my code that could cause this so I suspect a timing issue in the SDK code. It looks to me like the maybe background irq is being fired during the time that it is being removed although I really don't understand the SDK code in detail.

Stacktrace:
```
unhandled_user_irq_num_in_r0 () at /home/pi/pico/pico-sdk/src/rp2_common/pico_crt0/crt0.S:289
289         bkpt #0
(gdb) bt
#0  unhandled_user_irq_num_in_r0 () at /home/pi/pico/pico-sdk/src/rp2_common/pico_crt0/crt0.S:289
#1  <signal handler called>
#2  0x1000153c in set_raw_irq_handler_and_unlock (num=num@entry=13, handler=handler@entry=0x100001cd <isr_irq9>, save=save@entry=0)
    at /home/pi/pico/pico-sdk/src/rp2_common/hardware_irq/irq.c:59
#3  0x1000194c in irq_remove_handler (num=num@entry=13, handler=handler@entry=0x10013d01 <cyw43_gpio_irq_handler>)
    at /home/pi/pico/pico-sdk/src/rp2_common/hardware_irq/irq.c:575
#4  0x1000129e in gpio_remove_raw_irq_handler_masked (gpio_mask=gpio_mask@entry=16777216, handler=handler@entry=0x10013d01 <cyw43_gpio_irq_handler>)
    at /home/pi/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c:243
#5  0x10013d5c in gpio_remove_raw_irq_handler (handler=<optimized out>, gpio=24)
    at /home/pi/pico/pico-sdk/src/rp2_common/hardware_gpio/include/hardware/gpio.h:819
#6  cyw43_irq_deinit (param=<optimized out>) at /home/pi/pico/pico-sdk/src/rp2_common/pico_cyw43_driver/cyw43_driver.c:66
#7  0x10014174 in async_context_threadsafe_background_execute_sync (self_base=0x20002348 <cyw43_async_context_threadsafe_background>,
    func=0x10013d41 <cyw43_irq_deinit>, param=0x0)
    at /home/pi/pico/pico-sdk/src/rp2_common/pico_async_context/async_context_threadsafe_background.c:156
#8  0x10013e5e in async_context_execute_sync (param=0x0, func=<optimized out>, context=0x20002348 <cyw43_async_context_threadsafe_background>)
    at /home/pi/pico/pico-sdk/src/rp2_common/pico_async_context/include/pico/async_context.h:256
#9  cyw43_driver_deinit (context=context@entry=0x20002348 <cyw43_async_context_threadsafe_background>)
    at /home/pi/pico/pico-sdk/src/rp2_common/pico_cyw43_driver/cyw43_driver.c:121
#10 0x100103bc in cyw43_arch_deinit () at /home/pi/pico/pico-sdk/src/rp2_common/pico_cyw43_arch/cyw43_arch_threadsafe_background.c:70
#11 0x10000bc4 in WifiConnection::~WifiConnection (this=this@entry=0x20041a34, __in_chrg=<optimized out>)
    at /home/pi/dev/pico-weather-mqtt/wifi_connection.cpp:44
#12 0x1000043a in publish_sensor_values (values=...) at /usr/include/newlib/c++/12.2.1/bits/new_allocator.h:90
#13 0x10000766 in main () at /home/pi/dev/pico-weather-mqtt/uvsensor.cpp:138
```

Is this my code at fault or have I hit an SDK bug?

peterharperuk, 2025-01-13:
I feel like I've seen this before, but I can't reproduce it. I would guess it's an sdk bug. It strikes me that cyw43_irq_deinit is doing things in the wrong order. Are you able to try moving the call to cyw43_set_irq_enabled in cyw43_irq_deinit so it's called before gpio_remove_raw_irq_handler?

```
uint32_t cyw43_irq_deinit(__unused void *param) {
#ifndef NDEBUG
    assert(get_core_num() == async_context_core_num(cyw43_async_context));
#endif
    gpio_remove_raw_irq_handler(CYW43_PIN_WL_HOST_WAKE, cyw43_gpio_irq_handler);
    cyw43_set_irq_enabled(false); // MOVE THIS SO IT'S CALLED BEFORE gpio_remove_raw_irq_handler
    return 0;
}
```

gazzyt, 2025-01-13:
Yep - I've made that change and it's now running on the PICO. I'll update you in a couple of days unless it crashes sooner.

Thanks,
Gary

gazzyt, 2025-01-15:
It's been solid for a couple of days now so I reckon this was the fix. I have submitted a PR with the change.

Thanks for the help!

peterharperuk, 2025-01-15:
Nice!

lurch, 2025-01-22:
Linked PR has been merged into `develop`.

PR #2185 (2025-01-20): Disable irq before removing handler in cyw43_irq_deinit
This changes cyw43_irq_deinit to disable the irq before removing the handler otherwise the irq can occur after the handler has been removed causing the unhandled irq handler to be used thus hanging the device.

Fixes #2176
closed 2025-01-22 by lurch