raspberrypi/pico-sdk #2178: USB Buffer size

rapzak, 2025-01-13:
Make an option to increase buffersize from user code...

Std settings on CDC as today gives ~200kbytes/sec
with increased buffer gives ~600kbytes/sec

Maybe make it posible to define custom buffers, but also pre-define..  like (low mem - std as today - medium and max mem...)
I have not troed to increase further to see more impact yet...

My changes to archieve 600kbyte:

// CHANGE CHANGE CHANGE
//.pico-sdk\sdk\2.1.0\src\rp2_common\pico_stdio_usb\include\tusb_config.h
//#define CFG_TUD_CDC_RX_BUFSIZE  (256)
//#define CFG_TUD_CDC_TX_BUFSIZE  (256)
#define CFG_TUD_CDC_RX_BUFSIZE  1024// (TUD_OPT_HIGH_SPEED ? 512 : 64)
#define CFG_TUD_CDC_TX_BUFSIZE 1024//  (TUD_OPT_HIGH_SPEED ? 512 : 64)
// CDC Endpoint transfer buffer size, more is faster
#define CFG_TUD_CDC_EP_BUFSIZE  1024// (TUD_OPT_HIGH_SPEED ? 512 : 64)
// CHANGE CHANGE CHANGE END

I am not sure how to implement so it can be used from user code...

Thank you

rapzak, 2025-01-13:
The most important line:
#define CFG_TUD_CDC_EP_BUFSIZE 

I just tried with all 3 defines the same, and above 768 the increase in speed is close to 0....
But if the 3 defines can be differentiated i have not tested

kilograham, 2025-02-07:
the ability to override these is merged into develop (i.e. you can set them via `target_compile_definitions` in your CMake

lets leave having "presets" to a separate issue

closed 2025-02-07 by kilograham