raspberrypi/pico-sdk #2179: POWMAN Sleep Is Not Setting; Returns PICO_OK

dinesh-slx, 2025-01-13:
```
/*
 * SPDX-License-Identifier: BSD-3-Clause
 * (c) 2023 Raspberry Pi (Trading) Ltd.
 *
 * Example code demonstrating valid sleep/wakeup configuration
 */

#include <inttypes.h>
#include <stdint.h>
#include <stdio.h>

#include "hardware/gpio.h"
#include "hardware/powman.h"
#include "hardware/timer.h"
#include "pico.h"
#include "pico/stdlib.h"

#include "pico/btstack_cyw43.h" // If you are using Wi-Fi/BLE
#include "pico/cyw43_arch.h"    // If you are using Wi-Fi/BLE

// -----------------------------------------------------------------------------
// User Configuration
// -----------------------------------------------------------------------------

#define WAKE_PIN 16
#define LED_PIN CYW43_WL_GPIO_LED_PIN // Use the LED pin from CYW43
#define INACTIVITY_TIMEOUT_MS 5000    // Sleep after 5s of inactivity

// -----------------------------------------------------------------------------
// Forward Declarations
// -----------------------------------------------------------------------------
bool init_cyw43_peripherals(void);
void deinit_cyw43_peripherals(void);
void init_gpio(void);
powman_power_state configure_sleep_state(void);
bool configure_wakeup_state(powman_power_state sleep_state, powman_power_state wakeup_state);
void clear_wakeup_state(void);

// -----------------------------------------------------------------------------
// Initialize GPIOs
// -----------------------------------------------------------------------------
void init_gpio() {
  // Wake pin
  gpio_init(WAKE_PIN);
  gpio_set_dir(WAKE_PIN, GPIO_IN);
  gpio_pull_up(WAKE_PIN);

  // LED pin
  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);
}

// -----------------------------------------------------------------------------
// Initialize CYW43 (Wi-Fi/Bluetooth); adjust or remove if not needed
// -----------------------------------------------------------------------------
bool init_cyw43_peripherals() {
  if (cyw43_arch_init()) {
    printf("Failed to initialize cyw43_arch\n");
    return false;
  }
  return true;
}

// -----------------------------------------------------------------------------
// Deinitialize CYW43
// -----------------------------------------------------------------------------
void deinit_cyw43_peripherals() { cyw43_arch_deinit(); }

// -----------------------------------------------------------------------------
// Configure the sleep state
// IMPORTANT: Must turn ON Switched Core (bit 3) to have a valid wakeup path
// -----------------------------------------------------------------------------
powman_power_state configure_sleep_state() {
  powman_power_state sleep_state = POWMAN_POWER_STATE_NONE;

  // Turn on XIP cache (bit 2)
  sleep_state = powman_power_state_with_domain_on(sleep_state, POWMAN_POWER_DOMAIN_XIP_CACHE);
  printf("Sleep State after XIP_CACHE: 0x%08" PRIx32 "\n", sleep_state);

  // Turn on SRAM bank 1 (bit 0)
  printf("Get the powman state: %d\n", powman_get_power_state());
  sleep_state = powman_power_state_with_domain_on(sleep_state, POWMAN_POWER_DOMAIN_SRAM_BANK1);
  printf("Sleep State after SRAM_BANK1: 0x%08" PRIx32 "\n", sleep_state);

  // Turn on SRAM bank 0 (bit 1)
  sleep_state = powman_power_state_with_domain_on(sleep_state, POWMAN_POWER_DOMAIN_SRAM_BANK0);
  printf("Sleep State after SRAM_BANK0: 0x%08" PRIx32 "\n", sleep_state);

  // IMPORTANT: Turn on Switched Core (bit 3)
  sleep_state = powman_power_state_with_domain_off(sleep_state, POWMAN_POWER_DOMAIN_SWITCHED_CORE);
  printf("Sleep State after SWITCHED_CORE: 0x%08" PRIx32 "\n", sleep_state);

  return sleep_state;
}

// -----------------------------------------------------------------------------
// Clear all wakeup sources and re-enable the needed ones
// -----------------------------------------------------------------------------
void clear_wakeup_state() {
  // Disable all wake-up sources
  powman_disable_all_wakeups();

  // Example: Re-enable GPIO wake-up on WAKE_PIN
  //   - 1st parameter: wakeup_num index (0..3 typically)
  //   - 2nd parameter: gpio pin number
  //   - 3rd parameter: edge (true for edge, false for level)
  //   - 4th parameter: level (true for high, false for low)
  // If you want “any edge,” you might need to adapt logic or hardware design.
  powman_enable_gpio_wakeup(0, WAKE_PIN, false, false);
}

// -----------------------------------------------------------------------------
// Configure wakeup state so it matches the needed domains
// -----------------------------------------------------------------------------
bool configure_wakeup_state(powman_power_state sleep_state, powman_power_state wakeup_state) {

  wakeup_state = powman_power_state_with_domain_on(wakeup_state, POWMAN_POWER_DOMAIN_SWITCHED_CORE);
  wakeup_state = powman_power_state_with_domain_on(wakeup_state, POWMAN_POWER_DOMAIN_XIP_CACHE);
  printf("Wake up state: %d\n", wakeup_state);
  if (!powman_configure_wakeup_state(sleep_state, wakeup_state)) {
    printf("Invalid sleep->wakeup state combination!\n");
    return false;
  }
  return true;
}

// -----------------------------------------------------------------------------
// Main
// -----------------------------------------------------------------------------
int main() {
  // Initialize stdio
  stdio_init_all();
  sleep_ms(5000);
  // Initialize GPIO
  init_gpio();

  // (Optional) Initialize Wi-Fi/BLE
  if (!init_cyw43_peripherals()) {
    return -1;
  }

  // Blink LED briefly
  cyw43_arch_gpio_put(LED_PIN, true);
  sleep_ms(1000);
  cyw43_arch_gpio_put(LED_PIN, false);

  // Enable GPIO wakeup on WAKE_PIN
  powman_enable_gpio_wakeup(0, WAKE_PIN, false, false);

  // Configure the desired sleep state
  powman_power_state sleep_state = configure_sleep_state();

  // For demonstration, we’ll wake to the same state
  powman_power_state wakeup_state = sleep_state;

  // Validate sleep->wakeup transition
  if (!configure_wakeup_state(sleep_state, wakeup_state)) {
    // If invalid, bail out (or do something else)
    return -1;
  }

  printf("RP2350 is running. It will enter sleep after %u ms of inactivity.\n",
         INACTIVITY_TIMEOUT_MS);

  // Track last user activity
  absolute_time_t last_activity_time = get_absolute_time();

  while (true) {
    // Check for user input over USB (if connected)
    if (stdio_usb_connected()) {
      if (uart_is_readable(uart0)) {
        int c = getchar_timeout_us(0);
        if (c != PICO_ERROR_TIMEOUT) {
          printf("Received: %c\n", (char)c);
          last_activity_time = get_absolute_time(); // reset timer
        }
      }
    }

    // Check inactivity threshold
    if (absolute_time_diff_us(last_activity_time, get_absolute_time()) >
        (INACTIVITY_TIMEOUT_MS * 1000)) {
      printf("No activity detected for %u ms. Going to sleep.\n", INACTIVITY_TIMEOUT_MS);
      deinit_cyw43_peripherals();

      sleep_run_from_lposc();
      // Turn off LED
      cyw43_arch_gpio_put(LED_PIN, false);

      // Deinit Wi-Fi/BLE
      deinit_cyw43_peripherals();

      // Ensure wakeup state is clear and configured
      clear_wakeup_state();
      // Request the power state
      printf("Sleep_state: %d\n", sleep_state);
      int result = powman_set_power_state(sleep_state);
      printf("Result: %d\n", result);
      printf("Get the powman state: %d\n", powman_get_power_state());
      if (result != PICO_OK) {
        printf("Failed to set power state: %d\n", result);
        // Handle error (retry, etc.) or just loop forever
      }

      // Once set, the RP2350 sleeps and typically restarts on wakeup.
      // Code below here will not run until after a reset or re-init sequence.
    }
  }

  // Should never get here
  return 0;
}
```
When the RP2040 on the Pico 2W enters sleep for the first time, it achieves a power consumption of about 300 µA, which is expected. Upon pressing the wake-up button, the device wakes up as intended. However, after the inactivity timeout, it does not return to sleep mode as expected. Instead, the function powman_set_power_state simply returns PICO_OK without actually entering the low-power state.

On subsequent inactivity timeouts, the issue persists: the device does not go to sleep even though the function powman_set_power_state continues to return PICO_OK.

peterharperuk, 2025-01-13:
I think there are bugs in your code. I fiddled with it and it seems to work now. See the attached zip. 
[powman_test.zip](https://github.com/user-attachments/files/18398853/powman_test.zip)

closed 2025-01-13 by peterharperuk

dinesh-slx, 2025-01-14:
serial prints from  Pico 2W are not appearing after the device wakes from sleep

peterharperuk, 2025-01-14:
You mean from the uart? It was working for me.

dinesh-slx, 2025-01-14:
I'm not using UART; I'm using USB for my serial connection

peterharperuk, 2025-01-14:
Odd. Just tried it and USB stdio worked for me. I suggest you also use uart stdio for debug in case you have some other problem.

dinesh-slx, 2025-01-15:
Is there a way to know which GPIO pin woke the system from sleep?

lurch, 2025-01-15:
> Is there a way to know which GPIO pin woke the system from sleep?

It look like there's not (yet?) an SDK wrapper function for it, but a quick look at https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf shows that the `POWMAN` block has a `LAST_SWCORE_PWRUP` register which sounds like it might be what you're looking for? (in combination with the `PWRUP1`, `PWRUP2`, `PWRUP3` and `PWRUP4` registers)
See https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2350/hardware_regs/include/hardware/regs/powman.h#L1842