raspberrypi/pico-sdk #2181: Bluetooth stops working after (roughly)6..8 connections/disconnections.

rvt, 2025-01-13:
I have also mention the issue here : https://github.com/bluekitchen/btstack/issues/654#issuecomment-2585686002

----
I have been using sm_pairing_peripheral.c as my 'template' and use it to steam NMEA data from a ublox over BLE. Pairing and bonding works fine aswell as storing this on flash.
I am using this to steams data from an ublox (NMEA) to BLE, so it's a continues stream of NMEA data using notifications.

However, if I disconnect my device and reconnect a few times then after around 6 disconnects and re-connections the bluetooth stack stops responding and no more HCI events are received. No new connections are possible, no authentication is received and the log stays silent. Sometimes the stream might start for a second, then just stops. No disconnect events like ATT_EVENT_DISCONNECTED are received.

Other tasks the PICO is running keep running just fine for example reading serial data from a ublox etc, SPI interfaces and the WIFI web interface that the device is running.

From the wireshark I cannot see anything that stands out, but I am not well versed on bluetooth, or wireshark so I am not sure what to look at yet.

To Reproduce

Not sure yet what example I can use or what to use on the receiving device to reproduce. First I would like to get a feel of what os going on, and if others have seen this.

Expected behavior

I expect that I can always disconnect and reconnect any number of times.
----

Current BTstack branch: Develop
Bluetooth Controller: cyw43 on the PICO RP2040
Remote device: Xiaomi Redmi Note 10 but same issue with using an iPad or the lightblue application
FreeRTOS latest
PICO SDK 2.1.0

So far I have been trying to track it down into `cybt_hci_read(..) cybt_hci_read_packet (..)` where after the issue happens the driver keeps reading zero bytes, essentially showing something like the below over and over again. With a working situation it would just show up, when I disconnect I would not see these log messages over and over again.
Obviously, this could also be a logic error somewhere.

```
241.566: cybt_hci_read_packet: no data
241.572: cybt_hci_read: bt2host_in_val=1728 bt2host_out_val=1728 fw_b2h_buf_count=0
241.580: cybt_hci_read_packet: no data
241.961: cybt_hci_read: bt2host_in_val=1728 bt2host_out_val=1728 fw_b2h_buf_count=0
241.969: cybt_hci_read_packet: no data
241.976: cybt_hci_read: bt2host_in_val=1728 bt2host_out_val=1728 fw_b2h_buf_count=0
241.984: cybt_hci_read_packet: no data
242.344: cybt_hci_read: bt2host_in_val=1728 bt2host_out_val=1728 fw_b2h_buf_count=0
242.351: cybt_hci_read_packet: no data
242.745: cybt_hci_read: bt2host_in_val=1728 bt2host_out_val=1728 fw_b2h_buf_count=0
242.753: cybt_hci_read_packet: no data
242.759: cybt_hci_read: bt2host_in_val=1728 bt2host_out_val=1728 fw_b2h_buf_count=0
242.767: cybt_hci_read_packet: no data
243.141: cybt_hci_read: bt2host_in_val=1728 bt2host_out_val=1728 fw_b2h_buf_count=0
243.149: cybt_hci_read_packet: no data
243.157: cybt_hci_read: bt2host_in_val=1728 bt2host_out_val=1728 fw_b2h_buf_count=0
243.165: cybt_hci_read_packet: no data
243.354: cybt_hci_read: bt2host_in_val=1728 bt2host_out_val=1728 fw_b
```

After this keeps appearing no more data is read over bluetooth. Wifi keeps working as usually and my webinterface works as expected.

Anybody have seen this before?

Note: I am not using DMA in my application as noticed here : https://forums.raspberrypi.com/viewtopic.php?t=355701

peterharperuk, 2025-01-20:
Could you retest with the develop branch from today containing this fix? https://github.com/raspberrypi/pico-sdk/pull/2190

rvt, 2025-01-24:
hey @peterharperuk ,

my report seems to be an issue with the MTU from the ATT event rather that the device would not wake up.
However, I did experience some weird hangups on random time, not bluetooth related. I will try develop, but I can only provide anecdotal evidence since my issue is hard too reproduce. Sometimes I have a hang in 30 seconds, sometimes not a whole day.

peterharperuk, 2025-01-31:
@rvt It sounds like you fixed your issue? Is this still a problem?

rvt, 2025-02-01:
Hello @peterharperuk,

I have been using the patch and don't see any adverse effects. Hoeever, my issues was most lickly due to the MTU that was to high due to a configuration.

peterharperuk, 2025-02-03:
Ok, if there's no issue, I will close this.

closed 2025-02-03 by peterharperuk