raspberrypi/pico-sdk #2183: Ghost `USBCTRL_BASE` and colliding with `USBCTRL_DPRAM_BASE`

matsobdev, 2025-01-14:
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2350/hardware_regs/include/hardware/regs/addressmap.h#L96 and https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_regs/include/hardware/regs/addressmap.h#L72 doesn't belong anywhere in SDK and in https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2350/hardware_regs/RP2350.svd and https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_regs/RP2040.svd only `0x50100000` is mentioned as `USB_DPRAM`.

lurch, 2025-01-15:
I'm not sure if it answers your question, but see sections 2.2 and 4.1.2.7 in https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf , and sections 2.2 and 12.7.1.1.3 in https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf

matsobdev, 2025-01-16:
So perhaps it is to keep things in order.

closed 2025-01-16 by matsobdev