raspberrypi/pico-sdk #2189: Bug with clock division for CLK_SYS

cpottle9, 2025-01-17:
This almost certainly a hardware bug, not an SDK bug. 
I don't know where to submit hardware bugs.
I am testing with SDK 2.0.0. 

The field CLK_SYS_DIV_FRAC is not used  when CLK_SYS_DIV_INT is 1.
It is used for other values of CLK_SYS_DIV_INT.
So, it is impossible to divide the input clock by 1.5 but it is possible to divide by 2.5.

I work in micropython.
I made changes to the interpreter in C and also have micropython code I used for my testing.
If needed, I could reproduce the problem soley in C, but that might take awhile.

I am working to reduce power consumption for my application which runs on a PICO W.
Amongst other things, I want to disable pll_sys and clocking everything from pll_usb.
From my testing the cyw43 drivers don't like CPU frequency less than 64 MHZ.
I need to set pll_usb to a value that is an integer multiple of 48 MHZ.
I set pll_usb to run at 96 MHZ dividing by 2 for clk_usb, and clk_adc.
I want clk_sys to run at 64 MHZ the lowest frequency the cyw43 drivers tolerate.
I need to divide pll_usb by 1.5 to get 64 MHZ.

I configured pll_usb and clk_sys as follows:

```
  pll_init(pll_usb, 64*XOSC_HZ, 4, 2);
  clock_configure(clk_sys,
		  CLOCKS_CLK_SYS_CTRL_SRC_VALUE_CLKSRC_CLK_SYS_AUX,
		  CLOCKS_CLK_SYS_CTRL_AUXSRC_VALUE_CLKSRC_PLL_USB,
		  96*1000*1000,
		  64*1000*1000);
```
The clock_configure() work correctly. When I dump the clk_sys registers I get:
```
clk_sys:
   CTRL(0x00000021) src=AUXSRC(1) auxsrc=PLL_USB(1)
   DIV(0x00000180) int=0x000001(1) frac=0x80(128)
   SELECTED value=0x2
```

I used the FC0 frequency counter to measure the clock frequencies and I get:
```
src:    PLL_SYS  1 freq:       0.000 Khz
src:    PLL_USB  2 freq:   96000.000 Khz
src:    CLK_SYS  9 freq:   96000.000 Khz
```

Note CLK_SYS is 96 MHZ same as PLL_USB.

To further understand the problem I tried other values in CLK_SYS_DIV.
I tried CLK_SYS_DIV of 0x1c0 (divide by 1.75), 0x1f0(divide by 1.88 approximately) and the frequency counter still reports 96 MHZ for CLK_SYS.

I then tried 0x280(divide by 2.5) and it worked.
```
clk_sys:
   CTRL(0x00000021) src=AUXSRC(1) auxsrc=PLL_USB(1)
   DIV(0x00000280) int=0x000002(2) frac=0x80(128)
   SELECTED value=0x2
```

```
src:    PLL_SYS  1 freq:       0.000 Khz
src:    PLL_USB  2 freq:   96000.000 Khz
src:    CLK_SYS  9 freq:   38400.000 Khz
```

kilograham, 2025-01-17:
If you want suggestions on how to implement what you are trying to implement, I suggest using the forums https://forums.raspberrypi.com/viewforum.php?f=143

cpottle9, 2025-01-17:
My bad. I did not read that section at all.

I suggest a change to the register description in tables 224 and 235 for the FRAC field.
Make it clear the FRAC value is ignored when INT == 1.

kilograham, 2025-02-04:
docs updated in develop (and a minor bug fix)

PR #2225 (2025-02-04): clarify clock_configure and make it return the correct value achieved for bad clock inputs
fixes #2189
PR #2255 (2025-02-17): update previous fix as 1.0 < clkdiv < 2.0 IS supported on RP2350
fixes #2189 more accurately