raspberrypi/pico-sdk #2191: Pico don't reset automaticlly after programed on SDK 2.1.0

Temperature6, 2025-01-18:
With the same configure(cmake files, clion openocd download e.g.) ,pico will reset automaticlly after programed when I use sdk1.5.0, bug it doesn't work when I use SDK2.1.0

This is my pico_sdk_impot.cmake
```
# This is a copy of <PICO_SDK_PATH>/external/pico_sdk_import.cmake

# This can be dropped into an external project to help locate this SDK
# It should be include()ed prior to project()

#set(PICO_SDK_PATH "D:/EmbeddedDevelop/rpi_pico/sdk_151/pico-sdk")
set(PICO_SDK_PATH "D:/EmbeddedDevelop/rpi_pico/sdk_210/pico-sdk-2.1.0")

if (DEFINED ENV{PICO_SDK_PATH} AND (NOT PICO_SDK_PATH))
    set(PICO_SDK_PATH $ENV{PICO_SDK_PATH})
    message("Using PICO_SDK_PATH from environment ('${PICO_SDK_PATH}')")
endif ()

if (DEFINED ENV{PICO_SDK_FETCH_FROM_GIT} AND (NOT PICO_SDK_FETCH_FROM_GIT))
    set(PICO_SDK_FETCH_FROM_GIT $ENV{PICO_SDK_FETCH_FROM_GIT})
    message("Using PICO_SDK_FETCH_FROM_GIT from environment ('${PICO_SDK_FETCH_FROM_GIT}')")
endif ()
...........
```
I'd like to switch sdk version by comment and uncomment the "set" statement, 

And this is my cmakelist.txt
```
cmake_minimum_required(VERSION 3.21)

set(PICO_BOARD pico)
set(PICO_PLATFORM rp2040)

# initialize the SDK based on PICO_SDK_PATH
# note: this must happen before project()
include(pico_sdk_import.cmake)

project(rp2040_sdk210_test C CXX ASM)

set(CMAKE_C_STANDARD 11)

# Initialize the SDK
pico_sdk_init()

add_executable(${PROJECT_NAME} main.c)
target_link_libraries(${PROJECT_NAME} pico_stdlib)

# create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(${PROJECT_NAME})

```
This clion openocd download settings is unchanged when I switch sdk version
[image]

Am I forgetting what else I should configure？

Slion, 2025-01-18:
Same here after upgrading from 2.0.0 to 2.1.0.

Temperature6, 2025-01-19:
> Same issue observed on Windows 11 after upgrading from 2.0.0 to 2.1.0.
Yes, I'm also using Windows 11. If version 2.0.0 works well, I will give it a try.

Hanspeter7, 2025-02-04:
Same issue on debian bookworm after upgrading from 2.0.0 to 2.1.0.

atuleu, 2025-02-04:
I had some kind of similar issue, where my code won't fully reload on the two core. In my setup core1 is executing a task while core0 is waiting on it. When programming a new program, the core 1 never executes its code, I have to physically reboot the pico board to make it work.

I am using CMake/OpenOCD with a debug probe

I used to have the following OCD commands:
```cmake
add_custom_target(
  ${ARGS_TARGET}-upload
  COMMAND
    ${OPENOCD_EXECUTABLE} -s tcl -f interface/cmsis-dap.cfg #
      -c "adapter speed 5000" #
      -f target/rp2040.cfg #
      -c "program $<TARGET_FILE:${ARGS_TARGET}> verify reset exit"
  DEPENDS ${ARGS_TARGET}
)
```
And I have to split it up with a `reset init` after verification:

```cmake
add_custom_target(
  ${ARGS_TARGET}-upload
  COMMAND
    ${OPENOCD_EXECUTABLE} -s tcl -f interface/cmsis-dap.cfg #
      -c "adapter speed 5000" #
      -f target/rp2040.cfg #
      -c "program $<TARGET_FILE:${ARGS_TARGET}> verify" #
      -c "reset init" #
      -c "resume" # 
      -c "shutdown"
  DEPENDS ${ARGS_TARGET}
)
```
In the second version, the new program is loaded and run on both cores as expected.

I am not sure if it is completely related, but this issue appeared when switching to 2.1.0.

I am using `pico-sdk` `2.1.0`, and `openocd` `sdk-2.0.0`

kilograham, 2025-02-06:
> Same issue on debian bookworm after upgrading from 2.0.0 to 2.1.0.

same issue; i.e. with CLion, or not with CLion?

kilograham, 2025-02-06:
Also, is everyone sure they are deleting their build directory when switching SDK versions (CLion "Reset Cache and Reload Project" or better still just delete the directory, or create a new CMake config)

Slion, 2025-02-12:
On the SDK develop branch for now and this was working fine with `pico2_w`. Then it failed after switching to `pico_w` without deleting the build folder. Deleting the build folder after switching board fixed the issue however.

kilograham, 2025-02-13:
> On the SDK develop branch for now and this was working fine with `pico2_w`. Then it failed after switching to `pico_w` without deleting the build folder. Deleting the build folder after switching board fixed the issue however.

Yes, you cannot change platforms like this; i'm surprised the cmake configure didn't complain with a nice message telling you so

kilograham, 2025-03-29:
see also #2357 which was a bug preventing the build complaining if you switch PICO_PLATFORM

kilograham, 2025-03-29:
Frankly at this point I am not clear if this issue is a real problem; please weigh in if you still have issues (having deleted your build directory), or i will close

Slion, 2025-03-30:
@kilograham Just tested again and it works fine now. Feel free to close it.