raspberrypi/pico-sdk #2192: Board hangs after upgrading SDK from v2.0.0 to v2.1.0

Slion, 2025-01-18:
Using Pico W RP2040.
I have an app using btstack, tinyusb and lwip.
I was hoping to switch from Pico W to Pico 2 W but SDK v2.1.0 does not like me.
It hangs and I'm getting those sorts of logs:

```
[CYW43] do_ioctl(2, 263, 16): timeout
got unexpected packet 0
[CYW43] do_ioctl(0, 262, 9): timeout
Ignoring packet with wrong id 9 != 10
[CYW43] do_ioctl(0, 262, 9): timeout
Ignoring packet with wrong id 11 != 12
```

or 

```
[CYW43] do_ioctl(0, 262, 9): timeout
got unexpected packet 0
[CYW43] do_ioctl(0, 262, 9): timeout
got unexpected packet 0
[CYW43] do_ioctl(0, 262, 9): timeout
got unexpected packet 0
[CYW43] do_ioctl(0, 262, 9): timeout
got unexpected packet 0
[CYW43] do_ioctl(0, 262, 9): timeout
got unexpected packet 0
[CYW43] do_ioctl(0, 262, 9): timeout
got unexpected packet 0
```

I tried various combinations of tinyusb and other sub-modules without joy.
Also tried both cyw43-driver from 2.0.0 and 2.1.0 .
Notably tried tinyusb v0.18.0 as mentioned here: 
I looks like the issue lies with changes to the SDK itself.

peterharperuk, 2025-01-18:
@Slion could you try the develop branch? Is there any way we can reproduce the problem you're having?

Slion, 2025-01-18:
I sure could try the develop branch at some point.
I'm afraid it's a closed source project using multiple USB configurations and Bluetooth, no WiFi. Not sure when I'll have the time to try reproduce the issue outside the project.

peterharperuk, 2025-01-19:
Are you using freertos?
If not, which async method? Poll or background thread? 
Which btstack example would be closest to your app?
Do you have the same problem with rp2040 in the latest sdk?

Slion, 2025-01-19:
> Do you have the same problem with rp2040 in the latest sdk?

The issue is rp2040 Pico W on SDK 2.1.0.
No issue with rp2040 Pico W on SDK 2.0.0.

Slion, 2025-01-19:
> Which btstack example would be closest to your app?

HID host. Actually last night when trying the HOG host example I may have gotten a couple of freeze too. No logs though.

peterharperuk, 2025-01-19:
Ok, thanks. It might be related to https://github.com/raspberrypi/pico-sdk/issues/2186
I'll have to check what we did to change polling timing in the last release

Slion, 2025-01-19:
Looking at #2190 it seems you guys are still working out a proper fix for this issue right?

Slion, 2025-01-19:
If it's a polling you might be able to reproduce it using the HID host example and adding a high speed timer that blinks the LED, something like that:
```
    btstack_run_loop_set_timer_handler(&hid_timer, &hid_timer_callback);
    btstack_run_loop_set_timer(&hid_timer, 1);
    btstack_run_loop_add_timer(&hid_timer);
```

Maybe only blink every 250ms or something by checking the elapsed time in the callback.

peterharperuk, 2025-01-19:
Yes I still have some work to do on that issue. However if it's not too much trouble it would help a lot if you could tell us if the changes in that pr stops the problem happening.

peterharperuk, 2025-01-20:
@Slion Thanks for the HID host suggestion. I could reproduce the messages you were seeing and it does look like the same issue. I'll mark this as a duplicate of #2186 if you don't mind.

closed 2025-01-20 by peterharperuk

Slion, 2025-01-20:
@peterharperuk Glad you could reproduce it. I'm assuming you don't need me to test your fix with any urgency then.

Did you need to add the timer and LED blinking to the HID host example to reproduce it? Did you need to connect a Bluetooth device? Could you post the changes you made to the HID host example?

peterharperuk, 2025-01-20:
> I'm assuming you don't need me to test your fix with any urgency then.

Correct.

> Could you post the changes you made to the HID host example?

See the attached.

[0001-Test-for-Issue-2192.patch](https://github.com/user-attachments/files/18479467/0001-Test-for-Issue-2192.patch)