raspberrypi/pico-sdk #2193: Empty stdio_init_all when using usb stdio.

itewqq, 2025-01-18:
Hi there,

Just trying to build a usb printf example but it does not work. After I flash the firmware and re-connect the usb port, there is nothing happend, either in `dmesg` or under `/dev/tty*`. 

I just disassembled the elf and found that the `stdio_init_all` function is empty, pretty wired. If I set the `pico_enable_stdio_uart(test 1)` and there will be a function call to `stdio_uart_init` in `stdio_init_all` otherwise it's empty like this:

[image]

My gcc:
```shell
$ /usr/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 14.2.0
```

Pico SDK version: 2.1.0

test.c:
```
#include <stdio.h>
#include "pico/stdlib.h"

int main() {
    stdio_init_all();
    while (true) {
        printf("Hello, world!\n");
        sleep_ms(1000);
    }
}
```

CmakeList.txt:
```
# Generated Cmake Pico project file

cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)
set(PICO_SDK_PATH "~/pico/pico-sdk")
set(PICO_PLATFORM "rp2040")

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

project(test C CXX ASM)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

# Add executable. Default name is the project name, version 0.1

add_executable(test test.c )

pico_set_program_name(test "test")
pico_set_program_version(test "0.1")

pico_enable_stdio_uart(test 0)
pico_enable_stdio_usb(test 1)

# Add the standard library to the build
target_link_libraries(test pico_stdlib)

pico_add_extra_outputs(test)

```

build.sh:
```shell
#!/bin/bash

set -ex
cp ~/pico/pico-sdk/external/pico_sdk_import.cmake .
rm -rf build
mkdir build
cd build
cmake ..
make -j`nproc`
```

itewqq, 2025-01-18:
The firmware downloaded from https://rptl.io/pico-hello-usb works as expected, so I assume the hardware setting is good.

lurch, 2025-01-18:
Does the example at https://github.com/raspberrypi/pico-examples/tree/master/hello_world/usb work for you? See also Chapter 5 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

lurch, 2025-01-18:
Hmmm, did you forget to populate the `lib/tinyusb` submodule?

itewqq, 2025-01-20:
> Hmmm, did you forget to populate the `lib/tinyusb` submodule?

Thanks for the hint!
I didn't notice that the submodules should also be pulled... as the README.MD only mentioned that the master branch is what you need lol.

closed 2025-01-20 by itewqq