raspberrypi/pico-sdk #2194: Pico SDK 2.1.0 Error when Launch Debug tab is started

benji7480, 2025-01-18:
I am able to Build/Compile the example projects from the SDK 2.1.0
but after successful build, when launching debug tab i get the below shown error on the debug console.

Any help to overcome this error wud be good. Thanks

ERROR: Unable to start debugging. Unexpected GDB output from command "-exec-run". Don't know how to run.  Try "help target".

lurch, 2025-01-18:
ping @will-v-pi as this sounds like it might be the VSCode plugin.

benji7480, 2025-01-21:
> I'm not sure - sounds like they're using something else in VSCode rather than our extension, as that folder structure looks odd. I think we need more information to help.

Hello Will-v-Pi,

Here is the content of the VSCode extensions.json file below from one of the example projects:

{
    "recommendations": [
        "marus25.cortex-debug",
        "ms-vscode.cpptools",
        "ms-vscode.cpptools-extension-pack",
        "ms-vscode.vscode-serial-monitor",
        "raspberry-pi.raspberry-pi-pico"
    ]
}

let me know if u need any other info to figure out the problem.  any folder structure info? Thanks

will-v-pi, 2025-01-22:
It sounds like you're clicking the wrong button to debug your project - if you're using the Pico VS Code extension, then you need to use it's Debug Project button (see section 4.3 of the getting started guide) which uses Cortex-Debug. If you're still getting errors then please reply with a full set of steps to reproduce this issue.

will-v-pi, 2025-01-31:
@benji7480 Did clicking the correct button here solve your problems?

benji7480, 2025-02-01:
> [@benji7480](https://github.com/benji7480) Did clicking the correct button here solve your problems?

Thanks for asking me will-v-pi. Some of the example projects are working with the correct button. But, there are projects that dont work.. So i will soon post a project and steps that still lead me to that 1st error i reported...

benji7480, 2025-02-01:
the example project is hello_freertos. path of the project in my hard drive is  "C:\Users\vijay\Documents\pico-examples-master\freertos\hello_freertos"  when i try to compile it following error occurs...

"Configure failed. Would you like to attempt to configure with the CMake Debugger?"

by the way this project is not selectable on the new "Raspberry Pi Pico Project Tab" installed on the VS IDE along w/SDK2.1.0 to even try a 1st time build of the project. Kindly help

will-v-pi, 2025-02-01:
See raspberrypi/pico-vscode#81, the FreeRTOS example isn’t included yet.

You can’t just open a single example and build it unless you’ve modified the files correctly, you have to build the whole of pico-examples together as it’s one CMake project, which is why you’re getting that CMake error.

will-v-pi, 2025-02-03:
Closing this, as it looks like the issue has been trying to build individual pico examples as a project. If you want to build an individual example from pico-examples, you need to configure the whole project (ie pico-examples-master/), and then build the single example (eg `cmake --build --target hello_pwm` or `make hello_pwm`). You can't just configure the folder containg the example (eg pico-examples-master/hello_pwm/) as that is not a CMake project.

closed 2025-02-03 by will-v-pi