raspberrypi/pico-sdk #2195: SDK 2.1.0 : pio_get_dreq returning wrong value for pio2

UKTailwind, 2025-01-18:
As per the title. For all values of sm and tx/rx the values returned are those of PIO1

peterharperuk, 2025-01-24:
I just checked this. The wifi pio code goes to pio 2 by default. As far as I can see the dreq values are correct DREQ_PIO2_TX0 and DREQ_PIO2_RX0.

UKTailwind, 2025-01-24:
Definitely fails for me as used below together with my workround which clearly would cause problems if it was working

` 
       dma_channel_config c = dma_channel_get_default_config(dma_tx_chan);
       int pior=2, sm=0;
       PIO pio = (pior==0 ? pio0: (pior==1 ? pio1: pio2));
        channel_config_set_write_increment(&c, false);
        if(pior==2)channel_config_set_dreq(&c, 16+sm);
        else channel_config_set_dreq(&c, pio_get_dreq(pio, sm, true));
`

lurch, 2025-01-24:
> `PIO pio = (pior==0 ? pio0: (pior==1 ? pio1: pio2));`

This line seems to be duplicating the functionality of `pio_get_instance`? https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_pio/include/hardware/pio.h#L854

peterharperuk, 2025-01-24:
> Definitely fails for me as used below together with my workround which clearly would cause problems if it was working

Sorry, that sentence doesn't scan.

```
    int pior=2, sm=0;
    PIO pio = (pior==0 ? pio0: (pior==1 ? pio1: pio2));
    /*channel_config_set_write_increment(&c, false);
    if(pior==2)channel_config_set_dreq(&c, 16+sm);
    else channel_config_set_dreq(&c, pio_get_dreq(pio, sm, true));*/
    printf("%u\n", pio_get_dreq(pio, sm, true)); // <<-- Shows 16

```

peterharperuk, 2025-01-24:
I can't see how that's physically possible. One of us must be missing something.

kilograham, 2025-02-02:
moving to 2.2.0 as I can't see how this could be either

kilograham, 2026-09-30:
Closing as we haven't heard back.

closed 2026-09-30 by kilograham