raspberrypi/pico-sdk #2201: multicore_reset_core1 does not reset lockout_victim_initialized[1]

Gadgetoid, 2025-01-21:
@MichaelBell and I have been chasing a bug in MicroPython whereupon an attempt to write to the filesystem - after initialising core1 and performing a soft reset - would cause an indefinite lockup.

It looks like soft reset calls `multicore_reset_core1` which resets the core, but *does not* indicate that multicore lockout is no longer configured. This results in the following code  (in MicroPython's `rp2_flash.c`) blocking forever:

```
    if (multicore_lockout_victim_is_initialized(1 - get_core_num())) {
        multicore_lockout_start_blocking();
    }
```

I think `multicore_reset_core1` should include `lockout_victim_initialized[1] = false`.

*However* this runs counter to the comment (https://github.com/raspberrypi/pico-sdk/blob/95ea6acad131124694cda1c162c52cd30e0aece0/src/rp2_common/pico_multicore/multicore.c#L26-L32):

Could we at least get a `multicore_lockout_victim_deinit`?

Or, y'know, just unbreak this even if it might be broken in other cases? 😭

cdr60, 2025-01-21:
unsubscribe does not work anymore !

kilograham, 2025-02-03:
fix merged into develop

PR #2223 (2025-02-03): add multicore_lockout_victim_deinit. clear lockedout_flag for core1 during core1 reset
fixes #2201
closed 2025-02-03 by kilograham