raspberrypi/pico-sdk #2203: Hardware reset corrupts __uninitialized_ram() variable on Pico2 but not Pico

UKTailwind, 2025-01-21:
I save information in a variable defined as follows:
uint64_t __uninitialized_ram(_persistent);

On the RP2040 Pico this survives both a H/W reset (switch) and a watchdog reset
On the RP2350 it is corrupted by a H/W reset but works as expected with a watchdog reset

lurch, 2025-01-22:
See the comments in this issue: https://github.com/raspberrypi/pico-sdk/issues/2043

UKTailwind, 2025-01-22:
Sorry, but that didn't mean anything to me unless it simply confirms that the RP2350 overwrites RAM whereas the RP2040 doesn't

lurch, 2025-01-22:
> Sorry, but that didn't mean anything to me unless it simply confirms that the RP2350 overwrites RAM whereas the RP2040 doesn't.

Yes, exactly. "Ah, the problem is more fundamental than that: asserting the RUN pin powers down the memories, and they power back up when it's released." and "(The whole point of RUN is to be a global reset, and RAM being retained on RP2040 was just due to the lack of power gating on the RAMs.)".
So it's not so much "RP2350 overwrites RAM" but "RP2350 forgets the value of the RAM contents when you do a hardware reset".

UKTailwind, 2025-01-22:
>  asserting the RUN pin powers down the memories

I think that is a design mistake - certainly haven't seen it in other processors I've used STM32, PIC32 etc.

kilograham, 2026-10-17:
closing as now documented... that is the RP2350 behavior

closed 2026-10-17 by kilograham