raspberrypi/pico-sdk #2210: Unable to build tag 2.1.0 with, "fatal error: boot/bootrom_constants.h: No such file or directory"

lukebayes, 2025-01-24:
The picotool build works great for tag 2.0.0, and fails on tag 2.1.0 (and master).

It seems this directory, `src/rp2_common/boot_bootrom_headers/include` is not in the include path, but should be?

FWIW, I found this [issue/change](https://github.com/raspberrypi/pico-sdk/pull/1881), which looks similar.

Any tips appreciated. Thanks!

lukebayes, 2025-01-24:
More details:

```bash
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.1 LTS
Release:        24.04
Codename:       noble
```

kilograham, 2025-01-24:
You opened this in `pico-sdk` so I assume you are referring to building `picotool` as part of the SDK build, rather than building `picotool` separately. In either case i would make sure you have recreated your build directories

lurch, 2025-01-25:
> i would make sure you have recreated your build directories

I.e. make sure you delete the old build directory whenever you switch SDK versions.

will-v-pi, 2025-01-31:
Yes, this is the sort of error you get when building picotool 2.1.0 with SDK 2.0.0 or vice versa. I have a PR to throw an error for this in picotool which is awaiting review raspberrypi/picotool#189

As @kilograham and @lurch said, as long as you delete build directories before switching SDK version you should be fine.

will-v-pi, 2025-02-07:
This has been fixed in develop by merging https://github.com/raspberrypi/picotool/pull/189

closed 2025-02-07 by will-v-pi