raspberrypi/pico-sdk #2214: SDK 2.1.0: altcp_proxyconnect.c fails to compile on Debug builds

tjko, 2025-01-26:
When compiling "Debug" build (CMAKE_BUILD_TYPE=Debug), I noticed following error from LwIP:

```
/home/tjko/pico/pico-sdk-200/lib/lwip/src/apps/http/altcp_proxyconnect.c:580:5: error: initialization of 'void (*)(struct altcp_pcb *)' from incompatible pointer type 'enum tcp_state (*)(struct altcp_pcb *)' [-Wincompatible-pointer-types]
  580 |   , altcp_default_dbg_get_tcp_state
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/tjko/pico/pico-sdk-200/lib/lwip/src/apps/http/altcp_proxyconnect.c:580:5: note: (near initialization for 'altcp_proxyconnect_functions.keepalive_disable')
```

(not failing on "Release" builds as LWIP_DEBUG is not defined...)

peterharperuk, 2025-01-26:
Yes. It's an lwip bug

closed 2025-01-26 by peterharperuk

peterharperuk, 2025-01-27:
https://savannah.nongnu.org/bugs/index.php?62159

There are multiple patches that address it. Hopefully they'll release one of them soon.

tjko, 2025-01-28:
@peterharperuk, thanks.   I tested that patch and it seems to resolve the issue I encountered.