raspberrypi/pico-sdk #2216: USB low level example - fails on RP2350 with extra endpoint added

matsobdev, 2025-01-29:
Slightly modified (2 `BULK` endpoints - `0x01` and `0x81`) - [dev_lowlevel.zip](https://github.com/user-attachments/files/18588232/dev_lowlevel.zip), https://github.com/raspberrypi/pico-examples/tree/master/usb/device/dev_lowlevel works fine running on RP2040 and 2350. Adding extra `BULK IN` at `0x82` - [dev_lowlevel2.zip](https://github.com/user-attachments/files/18588336/dev_lowlevel2.zip) and RP2350 fails to enumerate, but RP2040 works fine. Changing compiler from `ARM GCC 14.2.1`/`13.3.1`/`9.3.1` to `ARM LLVM 18.1.3` and RP2350 works (`19.1.5` is not). `RISCV GCC 14.2.0` works as well. Release, debug, doesn't matter. Since example and its modification seems to be fine, so I guess it is not an example or SDK (I don't know is more precise :D) issue, so landed here.

matsobdev, 2025-01-31:
Some hint. Some HID devices with CherryUSB. Absolute mouse [working.zip](https://github.com/user-attachments/files/18624737/working.zip) and relative [working2.zip](https://github.com/user-attachments/files/18624739/working2.zip), totaling `HID_KEYBOARD_REPORT_DESC_SIZE` (inside `hid_keyboard_template.c`) at 164 bytes. Both work no matter compiler version. Now [compiler_dependent.zip](https://github.com/user-attachments/files/18624776/compiler_dependent.zip) has 163 bytes `HID_KEYBOARD_REPORT_DESC_SIZE`. It only works same like `dev_lowlevel`. So total descriptor size for compiler dependent is one byte smaller - no further changes (USB wise) apart user code (working reports with 6 bytes, compiler dependent with 4 - one byte smaller descriptor for 8 bit X and Y). RP2350.

jocover, 2025-03-13:
the bug is in the [memcpy((void *) ep->data_buffer, (void *) buf, len); ](https://github.com/raspberrypi/pico-examples/blob/84e8d489ca321a4be90ee49e36dc29e5c645da08/usb/device/dev_lowlevel/dev_lowlevel.c#L250), sometimes memcpy will freeze,I rewrote a simple memcpy function and it worked fine

```
void* my_memcpy(void *dst, const void *src, size_t n) {
    uint8_t *p = (uint8_t *)dst;
    uint8_t *q = (uint8_t *)src;
    
    size_t size = n;
    while (n--) *p++ = *q++;
    assert((uint64_t)p - (uint64_t)dst == size);
    return dst;
}
```

lurch, 2025-03-13:
@jocover Interesting... I wonder if the "which are different to most registers on RP2040" mentioned in section 4.1.2.7 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf might be relevant here? pinging @liamfraser again

liamfraser, 2025-03-13:
This is related to the change to Cortex M33 in RP2350. You can't do unaligned uint32_t accesses to "Device Memory" on Cortex M33 which is where the USB controller is located. We also had to write our own memcpy to work around this in the rp2040/rp2350 tinyusb driver.

https://github.com/hathach/tinyusb/blob/master/src/portable/raspberrypi/rp2040/rp2040_usb.c#L57

lurch, 2025-03-13:
@liamfraser Does that mean that https://github.com/raspberrypi/pico-examples/tree/master/usb/device/dev_lowlevel also needs to be updated to use a `unaligned_memcpy` function?

jocover, 2025-03-13:
> [@jocover](https://github.com/jocover) Interesting... I wonder if the "which are different to most registers on RP2040" mentioned in section 4.1.2.7 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf might be relevant here? pinging [@liamfraser](https://github.com/liamfraser) again
> 
> EDIT: Doh! I've just realised that this issue is about RP2350 rather than RP2040. Guess I need more coffee 😂

rp2040 uses mem_ops_aeabi.S, memcpy function from bootrom