raspberrypi/pico-sdk #2220: tools/build_all_headers.py exception - broken by rp2040 and rp2350 duplicate file names

d-c-d, 2025-01-30:
in the top level pico-sdk folder
```
$ python3 tools/build_all_headers.py  $PWD /tmp/build_all_headers.out
Traceback (most recent call last):
  File "/cygdrive/c/git/pico-sdk/tools/build_all_headers.py", line 66, in <module>
    raise Exception("Duplicate include file '{}' (found in both {} and {})".format(include_file, include_locations[include_file], include_path))
Exception: Duplicate include file 'boot_stage2/config.h' (found in both rp2040/boot_stage2/include and rp2350/boot_stage2/include)
```

workaround - add `IGNORE_DIRS.add('rp2040')` or IGNORE_DIRS.add('rp2350')

While useful, there were no examples of this tool being used.

lurch, 2025-01-30:
It's something I added a **long** time ago in #319 - it's not really something that we expected anybody to actually use for anything 😂

d-c-d, 2025-01-30:
For context, I only ran into the file `tools/build_all_headers.py` because I was trying to see if _semihosting_ would be valuable in my debug context, so I was looking at **all** ( of the few places semihosting was mentioned :-) )

d-c-d, 2025-02-16:
I'm guessing a milestone of none indicates a won't fix, is that correct?
What about deleting the file then?

lurch, 2025-02-18:
See #2292

lurch, 2025-03-18:
Fixed in `develop`

PR #2292 (2025-03-18): Fix tools/build_all_headers.py to cope with chip-specific headers
Fixes #2220 

With `SORT_HEADERS_BY_DIRECTORY = True` it shows you which directory the include files came from (which is the same behaviour as previous versions of this script) with the `#if PICO_RP2040` and `#if PICO_RP2350` headers at the very end
```c
// rp2_common/hardware_pio/include
#include "hardware/pio.h"
#include "hardware/pio_instructions.h"

// rp2_common/hardware_pll/include
#include "hardware/pll.h"

// rp2_common/hardware_powman/include
#include "hardware/powman.h"

// rp2_common/hardware_pwm/include
#include "har [... cut]
closed 2025-03-18 by lurch