raspberrypi/pico-sdk #2227: No CMAKE_CXX_COMPILER could be found mid-build when configuring pioasmBuild

Slion, 2025-02-03:
Using pico-sdk as a sub-module trying to switch to develop branch fails my build mid-flight somehow.
Currently using v2.0.0, tried v2.1.0 earlier this year but switched back to 2.0.0 because of runtime issues with btstack poll mode.
Windows 11.
I tried to explicitly declare CMAKE_CXX_COMPILER even though the compiler is found just fine during the project configuration but that did not help. Even setting the CXX environment did not help. 

```
[35/510] Performing configure step for 'pioasmBuild'
FAILED: pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/src/pioasmBuild-stamp/pioasmBuild-configure S:/Dev/Project/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/src/pioasmBuild-stamp/pioasmBuild-configure 
C:\WINDOWS\system32\cmd.exe /C "cd /D S:\Dev\GitHub\Slion\PicoHidProxy\build\pioasm && C:\Users\micro\.pico-sdk\cmake\v3.28.6\bin\cmake.exe --no-warn-unused-cli -DCMAKE_MAKE_PROGRAM:FILEPATH=C:/Users/micro/.pico-sdk/ninja/v1.12.1/ninja.exe -DPIOASM_FLAT_INSTALL=1 -DCMAKE_INSTALL_PREFIX=S:/Dev/Project/build/pioasm-install -DCMAKE_RULE_MESSAGES=OFF -DCMAKE_INSTALL_MESSAGE=NEVER -GNinja -CS:/Dev/Project/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/pioasmBuild-cache-Debug.cmake -S S:/Dev/Project/lib/pico-sdk/tools/pioasm -B S:/Dev/Project/build/pioasm && C:\Users\micro\.pico-sdk\cmake\v3.28.6\bin\cmake.exe -E touch S:/Dev/Project/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/src/pioasmBuild-stamp/pioasmBuild-configure"
Not searching for unused variables given on the command line.
loading initial cache file S:/Dev/Project/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/pioasmBuild-cache-Debug.cmake
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
```

will-v-pi, 2025-02-03:
I've confirmed what the issue is - the problem is caused because the develop SDK is versioned as 2.1.1, and therefore currently requires a pioasm version of 2.1.1. The extension only provides pioasm version 2.1.0, so the develop SDK tries to build pioasm and therefore runs into the issue of missing a compiler.

The simplest fix is to add `set(PICO_SDK_VERSION_REVISION 0)` after your `include(pico_sdk_import.cmake)`, which makes it think the version is 2.1.0, and therefore it uses the pre-compiled pioasm.

Slion, 2025-02-03:
Thanks, that did the trick.

closed 2025-02-03 by Slion