raspberrypi/pico-sdk #2234: increase XOSC delay multiplier

kilograham, 2025-02-04:
change this by default to 6

Wren6991, 2025-02-04:
Reasoning is:

* Quite a few third-party boards have increased this value from its default, suggesting that it is not a *good* default (https://github.com/search?q=repo%3Araspberrypi%2Fpico-sdk%20PICO_XOSC_STARTUP_DELAY_MULTIPLIER&type=code)
* Looking at crystal startup on a scope shows that the recommended AEL crystal with series resistor takes around 6 ms to reach full swing (measured by @P33M) so we have been living dangerously

The minimal change is just to bump the default provided here: https://github.com/raspberrypi/pico-sdk/blob/95ea6acad131124694cda1c162c52cd30e0aece0/src/rp2_common/hardware_xosc/include/hardware/xosc.h#L18

To avoid impact on boot time, we could also pull the XOSC enable earlier in boot (e.g. in crt0, before `.data` and `.bss` init) and then just poll during `xosc_init()`. However this will be an incompatible change for those who are overriding the weak `clocks_init()` symbol without setting `PICO_RUNTIME_SKIP_INIT_CLOCKS=1` etc.

Note `PICO_XOSC_STARTUP_DELAY_MULTIPLIER` is poorly named and is really just the startup delay in milliseconds.

dhalbert, 2025-02-04:
I was the instigator of this long ago via #457. We (Adafruit) had chosen slightly different xosc parts and were having longer startup problems on a few boards. We didn't know how you were testing the Pico boards yourself and whether you were discarding some with longer delays as "bad". So this could improve your yield 😄 .

The default of 6 would still be too low for us, so we'll keep it at 64 for our boards, based on our empirical testing from that time.

Renaming is fine with us, of course. When I came up with the name it didn't occur to me (or the reviewers) that it was just milliseconds.

P33M, 2025-02-04:
This is a board-dependent parameter and should be measured with appropriate test equipment. It is a function of the selected crystal, XOSC drive strength (but due to a hardware bug on RP2040, only the lowest one works), parallel loading capacitance, and series resistor.

Ramp times for Pico and Pico 2 are basically identical. Your mileage will vary.

kilograham, 2025-02-05:
ok, so we should change it to 6 for just pico and pico2?

P33M, 2025-02-05:
The ramp time is unlikely to be any faster on boards using our recommended crystal and layout. I would default to 6.

kilograham, 2025-02-05:
ok for RP2350 boards or all RP2040 boards too?

kilograham, 2025-02-12:
merged into develop

PR #2245 (2025-02-12): Increase deault PICO_CONFIG: PICO_XOSC_STARTUP_DELAY_MULTIPLIER to 6ms
fixes #2234 

Also changed the description to make MULTIPLIER make sense!
closed 2025-02-12 by kilograham