raspberrypi/pico-sdk #2236: SDK 2.1.0 tinyusb panics on composite usb audio and cdc acm device

not1337, 2025-02-04:
The tinyusb version pulled by SDK 2.1.0 is missing a fix causing panic when using a composite usb device (audio and cdc/acm).  The problem seems to be well known and fixed in current versions of tinyusb. See: https://github.com/hathach/tinyusb/pull/1802
A future SDK release needs an updated version of tinyusb.

As a temporary fix for everyone affected, replace the dcd_rp2040.c of the pulled tinyusb library with:

https://raw.githubusercontent.com/howard0su/tinyusb/cd7ee9639ee5ceddcdaecde5842c2ce97a56ce63/src/portable/raspberrypi/rp2040/dcd_rp2040.c

The above fixes the problem for me and the link is included for convenience to reduce search time for everyone affected.

kilograham, 2025-02-04:
Note that we don't want that particular fix, but we want the change that is being made in TinyUSB upstream (which may or may not be this one https://github.com/hathach/tinyusb/pull/2937 )

kilograham, 2025-02-04:
OP: is this a regression from earlier SDK, or was that also broken due to TinyUSB/rp2040 issues?

not1337, 2025-02-05:
Sorry, don't know. SDK 2.1.0 is the first version I'm using and I never used rp2040, only rp2350.

Slion, 2025-02-14:
Is this bug specific to rp2040 or does it affect rp2350 too?

not1337, 2025-02-14:
In my case rp2350a, I don't have any rp2040.

mryndzionek, 2025-07-29:
> [@not1337](https://github.com/not1337) this should be fixed in the latest version of the SDK (2.1.1) as we now run TinyUSB 0.18.0, which contains this fix here [hathach/tinyusb#2847](https://github.com/hathach/tinyusb/pull/2847) - can you confirm this works?

The issue is still there on `0.18.0`. At least one more PR is missing: https://github.com/hathach/tinyusb/pull/2937

not1337, 2025-07-29:
Unfortunately I can't test in the next few months as I'm busy with my job.