raspberrypi/pico-sdk #2240: XIP_MAIN_BASE seems like redundant/deprecated name, since we have XIP_BASE

d-c-d, 2025-02-06:
It appears that if two variables have the same value there should be guidance as to when each should be used.
There is no `XIP_MAIN_BASE` for the rp20350, so it would be helpful to document if it is deprecated, or just left around for the bootrom and the rp2040 `xip_alias_check_addr` function

in `src/rp2040/hardware_regs/include/hardware/regs/addressmap.h`
```
#define XIP_BASE _u(0x10000000)
#define XIP_MAIN_BASE _u(0x10000000)
```

kilograham, 2025-02-06:
@matiasilva did this change by mistake from RP2040 to RP2350?

kilograham, 2025-02-06:
ah actually i guess maybe it was vestigial in RP2040

lurch, 2025-02-10:
See also sections 2.2.2 in https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf and https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf
Interestingly, `XIP_MAIN_BASE` doesn't appear at all in either PDF.

Although I guess for consistency perhaps it's worth modifying `address_mapped.h` to use `XIP_BASE` instead of `XIP_MAIN_BASE` ?
```console
$ git grep XIP_MAIN_BASE
src/rp2040/hardware_regs/include/hardware/regs/addressmap.h:#define XIP_MAIN_BASE _u(0x10000000)
src/rp2_common/hardware_base/include/hardware/address_mapped.h:    valid_params_if(ADDRESS_ALIAS, rc >= XIP_MAIN_BASE && rc < XIP_NOALLOC_BASE);
```

matiasilva, 2025-02-13:
Internal filtering was added to remove `XIP_MAIN0_BASE` and `XIP_MAIN1_BASE` from RP2350, I imagine this was not added for RP2040. If we filter on RP2040, we'd need to replace any usage of it.

lurch, 2025-02-13:
> If we filter on RP2040, we'd need to replace any usage of it.

Perhaps we'd also need to add a `#define XIP_MAIN_BASE XIP_BASE` for backwards-compatibility, so that we don't accidentally break any 3rd-party code that might have been using `XIP_MAIN_BASE` ?

kilograham, 2025-02-13:
let's just comment the one in RP2040 to say it is vestigial; i'll take that

kilograham, 2025-02-13:
duh, that is auto-generated - i'll move this into a later release