raspberrypi/pico-sdk #2250: `printf()` taking up space in binary, even if `pico_enable_stdio_usb(project 0)` and `pico_enable_stdio_uart(project 0)`

matsobdev, 2025-02-09:
Having project using SDK 2.1.0 and defining:
```
target_compile_definitions(hog_kbd PRIVATE
	PICO_PANIC_FUNCTION=
	)
```
for maximum code size savings. But when there is `printf()` somewhere in the code, then binary size rapidly grows. User code without `printf()`:
```
Memory region         Used Size  Region Size  %age Used
             RAM:       17468 B       512 KB      3.33%
       SCRATCH_X:           0 B         4 KB      0.00%
       SCRATCH_Y:           0 B         4 KB      0.00%
```
and with `printf()`:
```
Memory region         Used Size  Region Size  %age Used
             RAM:       27456 B       512 KB      5.24%
       SCRATCH_X:           0 B         4 KB      0.00%
       SCRATCH_Y:           0 B         4 KB      0.00%
```
Having `pico_enable_stdio_usb(project 0)` and `pico_enable_stdio_uart(project 0)` doesn't help getting rid of `printf()like` code from `*.dis` file. Sample [project.zip](https://github.com/user-attachments/files/18725286/project.zip) doesn't use TinyUSB or other SDK lib, just SDK itself. File `/usb_config.h` has `printf()` commented out: `#define CONFIG_USB_PRINTF(...) /*printf(__VA_ARGS__)*/`. 10 KB might be not a lot, but `XIP_CAHCE` might be exercised unnecessarily.

kilograham, 2025-02-09:
If you don't need printf use `pico_set_printf_implementation(TARGET none)`

Note you can also try the currently-rather-obtuse `pico_minimize_runtime(TARGET ...)` CMake method

kilograham, 2025-02-09:
it would be interesting to know what is pulling in `printf` though; you can tell from the `.dis` - note with `pico_set_printf_implementation(TARGET none)` you can leave panic, but it'll use `puts` instead - though that is pointless here too if you don't want any output at all

closed 2025-02-09 by kilograham

matsobdev, 2025-02-10:
`pico_set_printf_implementation(TARGET none)` still leaves some `printf`, but it is better anyway (`PICO_PANIC_FUNCTION=` but `#define CONFIG_USB_PRINTF(...) printf(__VA_ARGS__)`:
```
Memory region         Used Size  Region Size  %age Used
             RAM:       19732 B       512 KB      3.76%
       SCRATCH_X:           0 B         4 KB      0.00%
       SCRATCH_Y:           0 B         4 KB      0.00%
```
There is some logging like inside `usbd_initialize` function and `printf` is still there with `pico_set_printf_implementation(TARGET none)`. `pico_minimize_runtime(TARGET ...)`, `pico_minimize_runtime(TARGET PRINTF)` throws out an error:
```
../main.c: In function 'main':
../main.c:36:9: error: implicit declaration of function 'add_repeating_timer_ms' [-Wimplicit-function-declaration]
   36 |         add_repeating_timer_ms(-10, repeating_timer_callback, NULL, &timer);
      |         ^~~~~~~~~~~~~~~~~~~~~~
```
Eradicating `printf` from code (here with that alias or macro) fully wipes any `printf` residues. And it applies as well for other cases where `printf` is introduced:
https://github.com/raspberrypi/pico-sdk/blob/95ea6acad131124694cda1c162c52cd30e0aece0/src/rp2_common/pico_cyw43_driver/cybt_shared_bus/cybt_shared_bus_driver.c#L26-L30

kilograham, 2025-02-10:
Yes `pico_set_printf_implementation(TARGET none)` just changes the implementation... if you have callers of printf, they remain

@peterharperuk not sure that printf-ing the BT firmware ID is the right thing to do just because it is a debug build. maybe guard it with something else. (also style perhaps define empty printf to be `((void)0)`

kilograham, 2025-02-10:
Yes `pico_minimize_runtime` is not quite right, however you could probably try

`pico_minimize_runtime(INCLUDE DEFAULT_ALARM_POOL)` since you appear to be using that