raspberrypi/pico-sdk #2256: How to avoid to build pioasm and picotool ?

yizi15, 2025-02-12:
OS: windows10
ide: vscode + cmake
When I use `2.0.0` and chip rp2040， My project build ok.
When I use  chip  rp2350, The target pioasm and picotool will be build and thow a exception.
I try use '2.1.0', no work.
The target pioasm and picotool  can built in msys2. How can I avoid to build pioasm and picotool whe build my project?

will-v-pi, 2025-02-12:
It doesn’t sound like they’re using the extension, as that doesn’t select between chips, you instead have to select the board (ie Pico, Pico 2).

> How can I avoid to build pioasm and picotool whe build my project?

The answer to this though is to use the pico-vscode extension, and if you’re having issues with the extension then file a bug report against that repo, and fill out the bug report template with all the details.

yizi15, 2025-02-12:
I not use Raspberry Pi Pico. the extension new project or import always downloads the sdk but never success. I think there is need some way to select the depends build path. I will try to use the  Raspberry Pi Pico extension .

lurch, 2025-02-12:
If you're not doing so already, make sure you delete any build-output directories when switching boards / platforms / sdk-versions.
And if you're able to copy'n'paste the actual error message you're getting (or take a screenshot), that'd be very helpful.

yizi15, 2025-02-12:
the build fail si a toolchain bug, I switch from visual Studio ti mingw is build success!

closed 2025-02-12 by yizi15