raspberrypi/pico-sdk #2263: btstack: le_device_db_info fails

Slion, 2025-02-13:
SDK 2.1.0 and develop
Board Pico 2 W - though possibly not board specific

To reproduce modify `hod_host_demo` by adding the following `CheckDeviceDatabase` function and call it from `BTSTACK_EVENT_STATE` event.
Make sure you connect to at least one BLE device and reboot the board.
Output shows something like:
```
BLE: Device DB count: 1
BLE: Device DB info: 0 - 00:00:00:00:00:00
```

Basically `le_device_db_info` fails to fetch any device info.

```
static void CheckDeviceDatabase()
{
    int count = le_device_db_count();
    printf("BLE: Device DB count: %d\n", count);
    bd_addr_t address;
    int addr_type;
    sm_key_t smKey;

    while (count>0)
    {
        count--;
        le_device_db_info(count,&addr_type,address,smKey);
        printf("BLE: Device DB info: %d - %s\n", count, bd_addr_to_str(address));
    }
}
```

I was able to debug that a bit: `le_device_db_tlv_fetch` returns false.

mringwal, 2025-02-16:
@Slion Please use le_device_db_max_count instead of le_device_db_count(). The devices are not kept from 0..count, so there's good chance that your device might be stored with a different index.

Slion, 2025-02-16:
Thanks, that did it.

closed 2025-02-16 by Slion