raspberrypi/pico-sdk #2272: lwIP, linker error with PPP enabled

TheRealZago, 2025-02-16:
When compiling a basic application with PPP enabled and support for authentication (PAP, CHAP and friends), the linker fails with the following error:
```
/Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/PicoModem.dir/Users/zago/repo/pico-sdk/lib/lwip/src/netif/ppp/magic.c.o: in function `magic_churnrand':
[build] /Users/zago/repo/pico-sdk/lib/lwip/src/netif/ppp/magic.c:118:(.text.magic_churnrand+0x30): undefined reference to `sys_jiffies'
```

My `lwipopts.h` is loosely based on the file provided by the [pico_w examples](https://github.com/raspberrypi/pico-examples/blob/b6ac07f1946271de2817f94d8ffc0425ecb7c2a9/pico_w/wifi/lwipopts_examples_common.h), with just a few additions:
```c
#define NO_SYS 1
// ... omissis ...
#define PPP_SUPPORT 1
#define PAP_SUPPORT 1
#define CHAP_SUPPORT 1
#define MSCHAP_SUPPORT 1
#define PPP_SERVER 1
```

"HAL" is being provided by linking against `pico_lwip_nosys`.

# Workaround

I've noticed `pico_lwip_nosys` already defines `sys_now()`, so I've exteded its job by adding my own library hook function in a dedicated source file.

```c
// @file libhooks.c

#include "lwip/sys.h"
#include "hardware/timer.h"

u32_t sys_jiffies(void)
{
    return time_us_32();
}
```

Should I open a pull request with a cleaner SDK-side correction?

kilograham, 2025-02-16:
sounds good

peterharperuk, 2025-02-16:
You might have to wrap it with #if NO_SYS as sys_jiffies is defined for free rtos? And maybe have another way of disabling it in case someone has their own implementation.

kilograham, 2025-02-16:
I think you can just put `__weak` on it; chances are that no one else has implemented it but still. this would go in already guarded code in lwip_nosys.c so should be fine

kilograham, 2025-02-17:
fix merged into develop

closed 2025-02-17 by kilograham