raspberrypi/pico-sdk #2280: Enable UART on QSPI pins

retsifp, 2025-02-16:
Hi,

I'm doing tests with the UART bootloader, which is described in the [datasheet](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf) of RP2350 (Chapter 5.8).
The pins SD2 and SD3 of QSPI are used as TX and RX by the bootloader. It works very good, but it would make sense to use the same pins in 'user mode' for communication with another controller. However it looks like the SDK lacks any functions to switch UART to bank1 pins?
So at the moment it's not possible to use the same pins as the bootloader uses in 'user mode', disregarding the possibility to manually set the registers to do so (but for that I'm too far away ...).

Would be great if there could be a solution in the long term :)

BR Thomas.

burtyb, 2025-02-16:
I think the example in https://github.com/raspberrypi/pico-examples/pull/571 does what you need? It uses qspi_gpio_set_function() to switch the pins to UART and then uses them to communicate with the RP2350 that booted it.

retsifp, 2025-02-17:
Wow, @burtyb thanks! :-) 
This exactly what I needed, and it works just great!

By the way, my code is here: https://codeberg.org/retsifp/rp2350_uart
Didn't know about this pullrequest and found another solution to embed the 'remote' firmware in the 'main' firmware.

lurch, 2025-02-17:
So can this issue be closed @retsifp , or is there still missing SDK functionality that you'd like to see added?

retsifp, 2025-02-17:
For me it's fine, can be closed if you want. 
I guess it would maybe make sense to have the possibility to set the QSPI pin functions with the SDK, but I don't know how much effort that would be :)
Maybe it's as easy as integrating [void qspi_gpio_set_function(...)](https://github.com/raspberrypi/pico-examples/pull/571/files#diff-01e7bccabcba6cd47102f5d3e20168672d4644a717c2eb2ff2902b99aae67276R35) in the SDK?

lurch, 2025-02-17:
I'll close this as a duplicate of #1799 then 🙂

kilograham, 2025-02-17:
I'll leave this around, just because i thought (and maybe this isn't the case) that it might require changes to our UART code to make this friendly - if this is just a IO and FUNCSEL setup issue, then I guess we can indeed close - i just haven't looked at it in detail

lurch, 2025-02-17:
> i just haven't looked at it in detail

Me neither 😂  Pinging @will-v-pi as he wrote the previously-mentioned https://github.com/raspberrypi/pico-examples/pull/571

will-v-pi, 2025-02-18:
Yeah, it was just an io/pads issue - although stdio_uart_init might need modifications too if we wanted to support that

Fantonrko, 2025-04-11:
> Wow, [@burtyb](https://github.com/burtyb) thanks! :-) This exactly what I needed, and it works just great!
> 
> By the way, my code is here: https://codeberg.org/retsifp/rp2350_uart Didn't know about this pullrequest and found another solution to embed the 'remote' firmware in the 'main' firmware.

Hey, I see ```// same pins as the UART bootloader is using
#define PIN_UART_TX 3
#define PIN_UART_RX 4```
But why is the 3 and 4 pins?

will-v-pi, 2025-09-04:
There is a known bug on that example which prevents RX from working - fixed by https://github.com/raspberrypi/pico-examples/pull/697