raspberrypi/pico-sdk #2282: Launching Debug Project w/SDK 2.1.0 for my project

benji7480, 2025-02-17:
I imported my freertos based project on the SDK 2.1.0 Installation and i am able to compile and build the project w/o any errors.

But, when i launch the project  via the new  "Raspberry Pi Pico Project" tab using the "debug project" tab inside the same it throws the below error:

File not found "executable": "c:/Users/vijay/Documents/pico-examples-master/FreeRTOS-SMP-Demos-main/FreeRTOS/Demo/CORTEX_M0+_RP2040/Standard1/build/example.elf"

But,actually it should find the right "main1_full.elf" file as per naming of the executable in my project in the same folder as displayed in the error above .i.e... "c:/Users/vijay/Documents/pico-examples-master/FreeRTOS-SMP-Demos-main/FreeRTOS/Demo/CORTEX_M0+_RP2040/Standard1/build/main1_full.elf

But, it is looking for an example.elf file instead of my executable which is available in the build folder after successful build of my project. the visual studio enviroment also prompts me to change my projects launch.json file. i am using the "Pico Debug (Cortex-Debug)" mode for debugging the project. Here is the my launch.json file too below. Help needed to modify it correctly to pick up the "main1_full.elf" executable file in my projects build folder.

`{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Pico Debug (Cortex-Debug)",
            "cwd": "${userHome}/.pico-sdk/openocd/0.12.0+dev/scripts",
            "executable": "${command:raspberry-pi-pico.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "serverpath": "${userHome}/.pico-sdk/openocd/0.12.0+dev/openocd.exe",
            "gdbPath": "${command:raspberry-pi-pico.getGDBPath}",
            "device": "${command:raspberry-pi-pico.getChipUppercase}",
            "configFiles": [
                "interface/cmsis-dap.cfg",
                "target/${command:raspberry-pi-pico.getTarget}.cfg"
            ],
            "svdFile": "${userHome}/.pico-sdk/sdk/2.1.0/src/${command:raspberry-pi-pico.getChip}/hardware_regs/${command:raspberry-pi-pico.getChipUppercase}.svd",
            "runToEntryPoint": "main",
            // Fix for no_flash binaries, where monitor reset halt doesn't do what is expected
            // Also works fine for flash binaries
            "overrideLaunchCommands": [
                "monitor reset init",
                "load \"${command:raspberry-pi-pico.launchTargetPath}\""
            ],
            "openOCDLaunchCommands": [
                "adapter speed 5000"
            ]
        },
        {
            "name": "Pico Debug (Cortex-Debug with external OpenOCD)",
            "cwd": "${workspaceRoot}",
            "executable": "${command:raspberry-pi-pico.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "external",
            "gdbTarget": "localhost:3333",
            "gdbPath": "${command:raspberry-pi-pico.getGDBPath}",
            "device": "${command:raspberry-pi-pico.getChipUppercase}",
            "svdFile": "${userHome}/.pico-sdk/sdk/2.1.0/src/${command:raspberry-pi-pico.getChip}/hardware_regs/${command:raspberry-pi-pico.getChipUppercase}.svd",
            "runToEntryPoint": "main",
            // Fix for no_flash binaries, where monitor reset halt doesn't do what is expected
            // Also works fine for flash binaries
            "overrideLaunchCommands": [
                "monitor reset init",
                "load \"${command:raspberry-pi-pico.launchTargetPath}\""
            ]
        },
        {
            "name": "Pico Debug (C++ Debugger)",
            "type": "cppdbg",
            "request": "launch",
            "cwd": "${workspaceRoot}",
            "program": "${command:raspberry-pi-pico.launchTargetPath}",
            "MIMode": "gdb",
            "miDebuggerPath": "${command:raspberry-pi-pico.getGDBPath}",
            "miDebuggerServerAddress": "localhost:3333",
            "debugServerPath": "${userHome}/.pico-sdk/openocd/0.12.0+dev/openocd.exe",
            "debugServerArgs": "-f interface/cmsis-dap.cfg -f target/${command:raspberry-pi-pico.getTarget}.cfg -c \"adapter speed 5000\"",
            "serverStarted": "Listening on port .* for gdb connections",
            "filterStderr": true,
            "hardwareBreakpoints": {
                "require": true,
                "limit": 4
            },
            "preLaunchTask": "Flash",
            "svdPath": "${userHome}/.pico-sdk/sdk/2.1.0/src/${command:raspberry-pi-pico.getChip}/hardware_regs/${command:raspberry-pi-pico.getChipUppercase}.svd"
        },
    ]
}
`

lurch, 2025-02-19:
@will-v-pi I guess this ought to be moved to https://github.com/raspberrypi/pico-vscode ?

will-v-pi, 2025-02-19:
Yes, although the solution in this case is you probably need to use the [CMake Tools Extension Integration](https://github.com/raspberrypi/pico-vscode?tab=readme-ov-file#cmake-tools-extension-integration), as it sounds like the executable has a different name from the project name.

Alternatively you can just change the project name to `main1_full` instead of `example`.

You should not need to modify the launch.json file.