raspberrypi/pico-sdk #2286: inconsistent use of PICO_RP2350B in boards/include/boards

d-c-d, 2025-02-17:
It appears to be a typo that `boards/include/boards/hellbender_0001.h` is selecting the RP2350 VARIANT
by defining
```
#define PICO_RP2350B 1
```

The other board files ( even the likely associated `src/boards/include/boards/hellbender_2350A_devboard.h` ) seem to use
```
#define PICO_RP2350A 1
```
to indicate that "the current board has an RP2350 in an A (30 GPIO) package"
while the following is used to select the 48 pin RP2350B
```
PICO_RP2350A 0
```
Perhaps @CBJamo can comment

lurch, 2025-02-17:
Oooh, well spotted! #2290 updates the script which checks board headers to look for `#define PICO_RP2350B`.

kilograham, 2025-03-19:
merged into develop

PR #2359 (2025-03-19): Update hellbender_0001.h
Replace the incorrect `#define PICO_RP2350B 1` with the correct `#define PICO_RP2350A 0`

Fixes #2286
commit 645b33c3 (2025-03-19): Update hellbender_0001.h (#2359)

Replace the incorrect `#define PICO_RP2350B 1` with the correct `#define PICO_RP2350A 0`

Fixes #2286
closed 2025-03-19 by kilograham