raspberrypi/pico-sdk #2294: RP2040 Not Booting After Power Cycle

joerossrobotics, 2025-02-18:
## Issue Summary

When compiling and flashing a program onto an RP2040 using the Raspberry Pi Pico SDK (any version), the program runs fine initially. However, if I remove and reconnect power, the board does not boot. The only way to get the program running again is to manually enter `boot-select mode` and use `picotool` to reboot the device.

Interestingly, when compiling the same program using PlatformIO, the issue does not occur—the board boots correctly after a power cycle. I am also only experiencing this on one of the boards I own. All the rest work fine. This initially led me to believe this was a hardware issue until I discovered flashing a program build with PlatformIO worked as expected.

Any guidance or insights would be greatly appreciated!

## Steps to Reproduce

- Compile and flash any program.
- Verify that the program runs as expected.
- Disconnect and reconnect power.
- Observe that the board does not boot.
- Manually enter `boot-select` mode and use `picotool` to reboot.
- The program starts running again.

## Expected Behavior

The RP2040 should boot automatically when power is restored, without requiring intervention.

## Observed Behavior

- The board does not boot automatically after power is cycled.
- It requires entering `boot-select` mode and manually rebooting using `picotool`.
- Compiling the same program using PlatformIO does not exhibit this issue.

## CMake Configuration

Here is the CMakeLists.txt used for compilation:

```cmake
cmake_minimum_required(VERSION 3.22)
include(external/pico-sdk/pico_sdk_init.cmake)

project(pico_test)

pico_sdk_init()

add_executable(${PROJECT_NAME}
    "src/main.cpp"
)

target_link_libraries(${PROJECT_NAME} pico_stdlib)

pico_enable_stdio_usb(${PROJECT_NAME} 1)
pico_enable_stdio_uart(${PROJECT_NAME} 0)
```

## PlatformIO Configuration (Working Case)

The following platformio.ini was used:

```ini
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
board = pico
framework = arduino
board_build.core = earlephilhower
```

burtyb, 2025-02-18:
I would try adding add_compile_definitions(PICO_XOSC_STARTUP_DELAY_MULTIPLIER=64) to your CMakeLists.txt file (from what I can see arduino-pico uses that by default).

joerossrobotics, 2025-02-18:
@burtyb Wow, that fixed the issue immediately, thank you! Do you have any idea what could maybe cause this to happen?

lurch, 2025-02-18:
@joerossrobotics In your initial post you mention "RP2040" but you fail to mention which actual board you're using. On some boards, the crystal oscillator (which is what XOSC stands for) takes longer to stabilise than on other boards:
```console
$ git grep "#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER" src/boards/include/boards/*.h
src/boards/include/boards/adafruit_feather_rp2040.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/adafruit_feather_rp2040_usb_host.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/adafruit_feather_rp2350.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/adafruit_itsybitsy_rp2040.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/adafruit_kb2040.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/adafruit_macropad_rp2040.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/adafruit_qtpy_rp2040.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/adafruit_trinkey_qt2040.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/archi.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/machdyne_werkzeug.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/metrotech_xerxes_rp2040.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 16
src/boards/include/boards/net8086_usb_interposer.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/nullbits_bit_c_pro.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
src/boards/include/boards/seeed_xiao_rp2040.h:#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
```
If you're using one of these boards, you should adjust your build to set `PICO_BOARD`, as described in section 6.2 of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf