raspberrypi/pico-sdk #2305: `_cyw43_write_reg` assigns to `last_size`/`_header`/`_backplane_window` variables which are never used

dpgeorge, 2025-02-19:
The cyw43 SPI transport `_cyw43_write_reg()` function has this bit of code in it:
```c
    if (fn == BACKPLANE_FUNCTION) {
        // In case of f1 overflow
        self->last_size = 8;
        self->last_header[0] = buf[0];
        self->last_header[1] = buf[1];
        self->last_backplane_window = self->cur_backplane_window;
    }
```
(`self` is a cyw43 driver instance.)

These `last_xxx` variables are never used by `cyw43-driver` because `F1_OVERFLOW_CHANGE` is permanently disabled.  And that feature even has comments that it never actually worked.

It would be good to get rid of these variables in the cyw43 driver data structure, to save a little bit of RAM and clean it up.  So I propose removing their use from this sdk.

Alternatively the above code could be guarded by a `#if F1_OVERFLOW_CHANGE`, but then that config option would need to be made public, probably named `CYW43_F1_OVERFLOW_CHANGE`.

peterharperuk, 2025-02-19:
Yes - we should get rid of all that junk.

dpgeorge, 2025-02-19:
> we should get rid of all that junk

Can I just remove all the `F1_OVERFLOW_CHANGE` code?

peterharperuk, 2025-02-19:
Yes please - I meant to raise a PR to remove it - but kept forgetting. It was originally a suggestion from Infineon I believe. I'm not sure it ever actually worked and we managed to fix the issue another way and have never used this code again.

dpgeorge, 2025-02-20:
> Yes please - I meant to raise a PR to remove it - but kept forgetting

OK, done in https://github.com/georgerobotics/cyw43-driver/pull/134

lurch, 2025-03-18:
Fixed in `develop`

PR #2306 (2025-03-18): Remove assignment of cyw43 last_size/header/backplane_window variables
They variables are never used by the cyw43-driver code, because `F1_OVERFLOW_CHANGE` is disabled.

Fixes #2305 (see that issue for more detailed discussion).
closed 2025-03-18 by lurch