raspberrypi/pico-sdk #2308: pico-sdk 2.1.0 Downloading Picotool runs always upon repeated CMake calls

sdbbs, 2025-02-20:
I noticed the same problem as in https://github.com/raspberrypi/pico-sdk/issues/2081 using pico-sdk 2.1.0.

Sometimes (like these days) I end up in a situation that where I exclusively work on CMakeLists.txt, and I have to run `cmake` multiple times - and that is how I managed up end up irritated with this: because *every* time I run CMake, I get this "Downloading Picotool"! (Even worse, I work with a structure with [multiple subfolders inside `./build`](https://stackoverflow.com/questions/10083427/can-cmake-generate-a-single-makefile-that-supports-both-debug-and-release/77246382#77246382), so *every* time I run `cmake` (even when I run that command repeatedly after each other), I have to see through multiple times "Downloading Picotool" !!

Also, now that I see "Downloading Picotool" which ends up in my project, I'd prefer using that downloaded `picotool` locally for my project.

I would argue that this is a documentation problem, because nowhere in https://github.com/raspberrypi/picotool/blob/develop/README.md#usage-by-the-raspberry-pi-pico-sdk (or #2081) is neither implied nor explicitly stated, that if I get "Downloading Picotool" during CMake configuration, and I choose to use that downloaded one in my project, *I have to build it myself manually*. Also, I'd like a method to have this build done automatically (have a comment on this at end)

-----

Here is an example to demonstrate (note, done in MINGW64 (MSYS2) `bash` shell of Windows 10):

* `main.c` right from the [README.md](https://github.com/raspberrypi/pico-sdk/blob/2.1.0/README.md)

```c
#include <stdio.h>
#include "pico/stdlib.h"

int main() {
    stdio_init_all();
    printf("Hello, world!\n");
    return 0;
}
```

* `CMakeLists.txt` also right from the [README.md](https://github.com/raspberrypi/pico-sdk/blob/2.1.0/README.md)

```cmake
cmake_minimum_required(VERSION 3.13)

# initialize the SDK directly
include($ENV{PICO_SDK_PATH}/pico_sdk_init.cmake)

project(my_project)

# initialize the Raspberry Pi Pico SDK
pico_sdk_init()

# rest of your project

add_executable(hello_world
    main.c
)

# Add pico_stdlib library which aggregates commonly used features
target_link_libraries(hello_world pico_stdlib)

# create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(hello_world)
```

I put these in `/tmp/test`, then I setup to run CMake:

```none
$ cd /tmp/test
$ tree .
.
├── CMakeLists.txt
└── main.c

1 directory, 2 files

$ mkdir build && cd build

$ /c/Program\ Files/CMake/bin/cmake.exe ../ -DCMAKE_BUILD_TYPE=Debug -G "MSYS Makefiles"
PICO_SDK_PATH is C:/src/pico-sdk
Defaulting platform (PICO_PLATFORM) to 'rp2040' since not specified.
Defaulting target board (PICO_BOARD) to 'pico' since not specified.
Using board configuration from C:/src/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
[... 228 lines cut ...]
TinyUSB available at C:/src/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
Compiling TinyUSB with CFG_TUSB_DEBUG=1
BTstack available at C:/src/pico-sdk/lib/btstack
cyw43-driver available at C:/src/pico-sdk/lib/cyw43-driver
lwIP available at C:/src/pico-sdk/lib/lwip
mbedtls available at C:/src/pico-sdk/lib/mbedtls
Using PICO_EXAMPLES_PATH from environment ('C:/src/pico-examples')
-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys64/tmp/test/build
```

... *finally*, there is no "Downloading Picotool" step anymore!

-----

So, in order to get the build process to use the `picotool` obtained by the "Downloading Picotool" CMake step, I had to build `picotool` myself, by manually issuing the `make picotoolBuild` command.

Now, I would prefer NOT having to do this manual build of `picotool` - instead, I'd prefer setting up something in my project's `CMakeLists.txt`, so that if `cmake` is ran from an empty `./build` subfolder (or generally, a build state which triggers the "Downloading Picotool" step), then `picotool` should be automatically built right after the "Downloading Picotool" step - so I never see (and wait for) this step again if I have to run `cmake` again afterwards.

I'm aware that "Downloading Picotool" comes from [Findpicotool.cmake](https://github.com/raspberrypi/pico-sdk/blob/2.1.0/tools/Findpicotool.cmake#L42), which is triggered by the CMake function `pico_init_picotool()` in [pico-sdk/tools/CMakeLists.txt](https://github.com/raspberrypi/pico-sdk/blob/2.1.0/tools/CMakeLists.txt#L103).

So, I thought that I could somehow force a call to `make picotoolBuild` from my own project's `CMakeLists.txt` at start (so all the other points in my `CMakeLists.txt` that would otherwise trigger "Downloading Picotool" would see the built `picotool`, and avoid running the download unnecessarily) - provided I call `pico_init_picotool()` beforehand. However, https://stackoverflow.com/questions/79452002/force-cmake-to-generate-build-files-makefile-at-arbitrary-point-in-cmakelists implies that such an approach would not really be possible.

So, what other options do I have, to setup the CMake configuration -- so that `picotool` gets automatically built, right after the very first time "Downloading Picotool" is triggered?

will-v-pi, 2025-02-20:
I think you’re misunderstanding how CMake works - the `cmake` command itself does not do any building, it’s only the subsequent `make` command that actually does the build. The `cmake` command just configures all the directories and files so that the build tool (make/ninja/etc) can then build them. This is just how CMake works.

You do not need to manually build picotool using the picotoolBuild target - it will be automatically built as a dependency when you build your SDK project (ie when you run `make`). You will continue to get the “Downloading Picotool” message until you do this, but it will go away the first time you run `make`. This is the documented method of building the SDK, so I don’t think there is a documentation issue there.

Alternatively, if you just want to run `cmake` a lot without running `make`, you can just pass `-DPICO_NO_PICOTOOL=1` until your final `cmake` command where you can pass `-DPICO_NO_PICOTOOL=0`. This is not recommended, but should work fine if you’re insisting on not running `make` until the end.

The best solution would be to follow the advice in the warning and build & install picotool separately systemwide - that way you’ll never be annoyed by “Downloading Picotool” again.

sdbbs, 2025-02-20:
Thanks, @will-v-pi : 

> I think you’re misunderstanding how CMake works - the `cmake` command itself does not do any building, it’s only the subsequent `make` command that actually does the build. The `cmake` command just [...]

I think my misunderstanding comes from not applying the configuration nature of CMake to build dependencies (and I consider picotool in this case to be a build dependency); otherwise I was of course aware that CMake only does configuration on the level of a project - and then I'd have to call the build tool, here `make`, to actually build the project.

I guess being in a situation of calling CMake multiple times in a row (without calling `make` in between), and seeing "Downloading Picotool" multiple times per CMake call (due to my own modifications of `CMakeLists.txt`), made me lose perspective that the configuration nature of CMake applies to not just project dependencies, but also to build dependencies such as related tools.

> You do not need to manually build picotool using the picotoolBuild target - it will be automatically built as a dependency when you build your SDK project (ie when you run `make`). 

Yeah - for some reason I lost this from sight, in spite of this not being the first time I compile a pico-sdk project with CMake.

> You will continue to get the “Downloading Picotool” message until you do this, but it will go away the first time you run `make`. This is the documented method of building the SDK, so I don’t think there is a documentation issue there.

Ok, I have to agree with this; I guess I only noticed this to begin with because of being put in a situation to run CMake multiple times in a row (to fix bugs in my modifications of `CMakeLists.txt`), without running `make` in between.

> Alternatively, if you just want to run `cmake` a lot without running `make`, you can just pass `-DPICO_NO_PICOTOOL=1` until your final `cmake` command where you can pass `-DPICO_NO_PICOTOOL=0`. This is not recommended, but should work fine if you’re insisting on not running `make` until the end.

Nice - this seems like a decent workaround, if you need to run CMake multiple times (for instance to fix bugs or inspect variable values in a `CMakeLists.txt`) without running `make`

> The best solution would be to follow the advice in the warning and build & install picotool separately systemwide - that way you’ll never be annoyed by “Downloading Picotool” again.

I guess I thought that having separate versions of picotool for each project appealed to me, because projects would appear more isolated from each other, and plus I try to "install" source projects as rarely as possible (I either use package manager, or if building some tool from source, I try to call it with absolute path with the possible LD_LIBRARY_PATH and such - not sure where I picked up that habit)

Anyways, thanks for helping me put things in perspective - I guess now I realize I don't really have much to ask for in this case, so I'll be closing this.

closed 2025-02-20 by sdbbs

sdbbs, 2025-02-20:
I also arrived at a `CMakeLists.txt` setup to "prebuild" `picotool` at start of the `CMakeLists.txt`, so that later code in `CMakeLists.txt` considers `picotool` to be found, and it does not initiate "Downloading Picotool". 

This is done by creating a "dummy" directory with `CMakeLists.txt` project in `./build`, before any `CMakeCache.txt` is generated by the real project; building that via `execute_process` (so `cmake` inside that process is effectively a child of / separate from the "real" one), and finally cleaning up the `CMakeCache.txt` so it does not interfere with the configuration initiated by the remainder of the "real"/"master" `CMakeLists.txt`. As a consequence of this:

* The final `Makefile` will not have a `picotoolBuild` target anymore
* The `cmake` picotool prebuild works only if you start from an empty `./build` subfolder; otherwise, if you run `cmake` again, the prebuild code will fail with '`CMake Error: The source "...CMakeLists.txt" does not match the source "(other)...CMakeLists.txt" used to generate cache.`' (but that's OK for me, at least that does not interfere with the rest of the "real" CMakeLists.txt functionality).

Otherwise, I think it works, because in the end, a `.uf2` file does get generated.

Here is modified `CMakeLists.txt` (relative to the project given in OP):

```cmake
cmake_minimum_required(VERSION 3.13)

message("\n### prebuild picotool vvvvvvvvvv\n")
# https://crascit.com/2022/01/25/quoting-in-cmake/
file(WRITE ${CMAKE_BINARY_DIR}/pbptest/CMakeLists.txt [=[
cmake_minimum_required(VERSION 3.13...3.27)
include($ENV{PICO_SDK_PATH}/pico_sdk_init.cmake)
project(pbptest)
pico_sdk_init()
pico_init_picotool()
]=])
file(TOUCH ${CMAKE_BINARY_DIR}/pbptest/main.c)
message("Created ${CMAKE_BINARY_DIR}/pbptest/{CMakeLists.txt,main.c}")
message("Running `cmake -S pbptest -B . -G '${CMAKE_GENERATOR}'` in ${CMAKE_BINARY_DIR}")
execute_process(
  COMMAND ${CMAKE_COMMAND} -S pbptest -B . -G ${CMAKE_GENERATOR}
  WORKING_DIRECTORY ${CMAKE_BINARY_DIR}
)
message("Running `make picotoolBuild` in ${CMAKE_BINARY_DIR}")
execute_process(
  COMMAND ${CMAKE_COMMAND} --build . --target picotoolBuild
  WORKING_DIRECTORY ${CMAKE_BINARY_DIR}
)
execute_process(
  COMMAND find . -name picotool*
  WORKING_DIRECTORY ${CMAKE_BINARY_DIR}
)
message("Running `rm CMakeCache.txt` in ${CMAKE_BINARY_DIR} (for rest of original CMakeLists.txt)")
file(REMOVE "${CMAKE_BINARY_DIR}/CMakeCache.txt")
message("Running `rm -rf pbptest` in ${CMAKE_BINARY_DIR} (cleanup)")
file(REMOVE_RECURSE "${CMAKE_BINARY_DIR}/pbptest")
message("\n### end prebuild picotool ^^^^^^\n")

# initialize the SDK directly
include($ENV{PICO_SDK_PATH}/pico_sdk_init.cmake)

project(my_project)

# initialize the Raspberry Pi Pico SDK
pico_sdk_init()

# rest of your project

add_executable(hello_world
    main.c
)

# Add pico_stdlib library which aggregates commonly used features
target_link_libraries(hello_world pico_stdlib)

# create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(hello_world)
```

And here is the output:

<details>

[... 258 lines cut ...]
PICO_SDK_PATH is C:/src/pico-sdk
Target board (PICO_BOARD) is 'pico'.
Using board configuration from C:/src/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
Build type is Debug
Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
TinyUSB available at C:/src/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
Compiling TinyUSB with CFG_TUSB_DEBUG=1
BTstack available at C:/src/pico-sdk/lib/btstack
cyw43-driver available at C:/src/pico-sdk/lib/cyw43-driver
lwIP available at C:/src/pico-sdk/lib/lwip
mbedtls available at C:/src/pico-sdk/lib/mbedtls
Using PICO_EXAMPLES_PATH from environment ('C:/src/pico-examples')
-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys64/tmp/test/build

$ ls
_deps/       picotool/            CMakeCache.txt              hello_world.dis      hello_world.uf2
CMakeFiles/  pioasm/              CMakeDoxyfile.in            hello_world.elf      Makefile
generated/   pioasm-install/      CMakeDoxygenDefaults.cmake  hello_world.elf.map  pico_flash_region.ld
pico-sdk/    cmake_install.cmake  hello_world.bin             hello_world.hex
```

</details>

sdbbs, 2025-02-20:
Thanks @lurch 

> you also might be interested in using the `picotool_DIR` option

I had already seen `picotool_DIR`, but it kept confusing me, because I saw https://github.com/raspberrypi/pico-sdk/blob/2.1.0/tools/CMakeLists.txt#L120 :

```
set(picotool_DIR ${picotool_INSTALL_DIR}/picotool)
```

... and I couldn't parse - is that `/picotool` appended the executable file, or the directory?

Finally got that resolved after perceiving https://github.com/raspberrypi/pico-sdk/blob/2.1.0/tools/Findpicotool.cmake#L78

```
set(picotool_EXECUTABLE ${picotool_INSTALL_DIR}/picotool/picotool)
```

so, indeed, `picotool_DIR` is expected to have a "leaf" directory also called `picotool` (and that `/picotool` at end of `picotool_DIR` is intended to represent that "leaf" directory, not the executable)

lurch, 2025-02-20:
https://github.com/raspberrypi/picotool?tab=readme-ov-file#usage-by-the-raspberry-pi-pico-sdk 😉 
(I forgot to provide this link in my previous comment)