raspberrypi/pico-sdk #2309: Missing return values in new {disable/enable}_interrupts routines

earlephilhower, 2025-02-20:
SDK 2.1.1 included two new IRQ functions in `sync.h`.  They're defined as returning a value, but the code doesn't seem to actually return anything.  This will (likely) cause a crash when run with newer GCCs and throws a warning/error.

https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/hardware_sync/include/hardware/sync.h#L204-L212

https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/hardware_sync/include/hardware/sync.h#L217-L225

earlephilhower, 2025-02-20:
````
/home/runner/arduino_ide/hardware/pico/rp2040//pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h: In function 'uint32_t disable_interrupts()':
/home/runner/arduino_ide/hardware/pico/rp2040//pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h:212:1: error: no return statement in function returning non-void [-Werror=return-type]
  212 | }
      | ^
````

kilograham, 2025-03-09:
merged into develop

closed 2025-03-09 by kilograham