raspberrypi/pico-sdk #2313: BLE extended advertising not working with a Pi Pico 2 W

bittailor, 2025-02-23:
I wanted to create a BLE peripheral that uses extended advertising to send lager advertisements than the limits of legacy advertising.

It looks like this does _not work_ for Pi Pico 2 W. A minimal example where I try to set up a BLE peripheral using extended advertising can be seen [here](https://github.com/bittailor/bt-pico-ble-advertising/blob/main/extended_advertising/src/main.cpp). `ENABLE_LE_EXTENDED_ADVERTISING` is set in the [`btstack_config.h`](https://github.com/bittailor/bt-pico-ble-advertising/blob/b8ffe7e1635b3682cf77b78753c77fa990f9170f/extended_advertising/src/btstack_config.h#L20) and none of the `gap_extended_advertising...()` functions retuned an error during the [setup of the extended advertising](https://github.com/bittailor/bt-pico-ble-advertising/blob/b8ffe7e1635b3682cf77b78753c77fa990f9170f/extended_advertising/src/main.cpp#L49-L74). 

When digging deeper, I noticed that the [`hci_le_extended_advertising_supported`](https://github.com/bluekitchen/btstack/blob/2b49e57bd1fae85ac32ac1f41cdb7c794de335f6/src/hci.c#L1495-L1497)  method of BtStack returns false and with this it looks like the extended advertising is _silently just ignored_.

I then enabled the BtStack logging, still did not see an error or warning in regard to the extended advertising, but at startup it _dumps_ the supported local HCI commands  

```
[00:00:02.048] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_READ_REMOTE_EXTENDED_FEATURES (0) supported 2/6
[00:00:02.058] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_WRITE_SYNCHRONOUS_FLOW_CONTROL_ENABLE (1) supported 10/4
[00:00:02.069] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_READ_BUFFER_SIZE (2) supported 14/7
[00:00:02.078] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_WRITE_DEFAULT_ERRONEOUS_DATA_REPORTING (3) supported 18/3
[00:00:02.089] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_READ_ENCRYPTION_KEY_SIZE (4) supported 20/4
[00:00:02.098] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_SET_EVENT_MASK_PAGE_2 (5) supported 22/2
[00:00:02.108] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_WRITE_LE_HOST_SUPPORTED (6) supported 24/6
[00:00:02.117] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_LE_READ_REMOTE_FEATURES (7) supported 27/5
[00:00:02.127] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_REMOTE_OOB_EXTENDED_DATA_REQUEST_REPLY (8) supported 32/1
[00:00:02.137] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_WRITE_SECURE_CONNECTIONS_HOST (9) supported 32/3
[00:00:02.148] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_READ_LOCAL_OOB_EXTENDED_DATA_COMMAND (10) supported 32/6
[00:00:02.158] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_LE_WRITE_SUGGESTED_DEFAULT_DATA_LENGTH (11) supported 34/0
[00:00:02.169] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_LE_SET_ADDRESS_RESOLUTION_ENABLE (12) supported 35/1
[00:00:02.180] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_LE_READ_MAXIMUM_DATA_LENGTH (13) supported 35/3
[00:00:02.190] LOG -- hci.c.2793: Command SUPPORTED_HCI_COMMAND_LE_SET_HOST_FEATURE_V1 (17) supported 44/1
[00:00:02.199] LOG -- hci.c.2800: Local supported commands summary 00023fff
```

To my suprise the [`SUPPORTED_HCI_COMMAND_LE_SET_EXTENDED_ADVERTISING_ENABLE`](https://github.com/bluekitchen/btstack/blob/2b49e57bd1fae85ac32ac1f41cdb7c794de335f6/src/hci.c#L178) is not in this supported commands dump. This as I understood that extended advertising was added with Bluetooth 5 and [Pico 2 W _claims_ Bluetooth 5.2](https://www.raspberrypi.com/products/raspberry-pi-pico-2/):

> Raspberry Pi Pico 2 W features 2.4GHz 802.11n wireless LAN and Bluetooth 5.2, giving you even more flexibility in your IoT or smart product designs and expanding the possibilities for your projects.

Infineon states even Bluetooth® 5.4 for the [CYW43439](https://www.infineon.com/cms/en/product/wireless-connectivity/airoc-wi-fi-plus-bluetooth-combos/wi-fi-4-802.11n/cyw43439/)
>Infineon’s AIROC™ CYW43439 single-chip combo device features 1x1 single-band 2.4 GHz Wi-Fi 4 (802.11n) and Bluetooth® 5.4 is a reliable entry-level solution for low-cost applications. 

So I wonder is this an issue of BtStack or is it an issue of the included CYW43439 cyw43_driver/firmware? 

Or is it really not possible to use extended advertising with Pico 2 W hardware (released ~8 years after the release of extended advertising)?   

---

**HW & SW Versions**
- [Raspberry Pi Pico 2 W](https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#pico2w-technical-specification) 
- [pico-sdk 2.1.1](https://github.com/raspberrypi/pico-sdk/releases/tag/2.1.1): 
- [btstack 1.6.2](https://github.com/bluekitchen/btstack/releases/tag/v1.6.2)

matsobdev, 2025-02-23:
https://forums.raspberrypi.com/viewtopic.php?p=2235712&hilit=computer+says#p2235712

```
07/10/24 08:36:16.624  COM23@115200 c> Reset 
	  HCI Command
	COM23@115200
	[03 0C 00 ]
	opcode = 0x0C03 (3075, "Reset")

07/10/24 08:36:16.630  COM23@115200 <c Reset
	  HCI Command Complete Event
	COM23@115200
	[0E 04 ]: 01 03 0C 00 
	event = 0x0E (14,"Command Complete")
	Num_HCI_Command_Packets = 0x1 (1)
	Command_Opcode = 0xC03 (3075, "Reset")
	Status = 0x0 (0, "Success")

07/10/24 08:38:34.440  COM23@115200 c> LE_Set_Extended_Advertising_Enable 
	  HCI Command
	COM23@115200
	[39 20 06 ]: 01 01 00 00 00 00
	opcode = 0x2039 (8249, "LE_Set_Extended_Advertising_Enable")
	Enable = 0x1 (1, "Advertising is enabled")
	Number_of_Sets = 0x1 (1)
	Advertising_Handle[0] = 0x0 (0)
	Duration[0] = 0x00 (0)
	Max_Extended_Advertising_Events[0] = 0x0 (0)

07/10/24 08:38:34.442  COM23@115200 <c LE_Set_Extended_Advertising_Enable
	  HCI Command Complete Event
	COM23@115200
	[0E 04 ]: 01 39 20 01 
	event = 0x0E (14,"Command Complete")
	Num_HCI_Command_Packets = 0x1 (1)
	Command_Opcode = 0x2039 (8249, "LE_Set_Extended_Advertising_Enable")
	Status = 0x1 (1, "Unknown HCI Command")
```

Since `cyw43_driver` just forwards from `CYW43439`, I bet it just doesn't recognise for example `LE_Set_Extended_Advertising_Enable` command, thus extended advertising is not supported (it is an optional feature).

bittailor, 2025-02-24:
I see, so I was misled by the _Bluetooth 5.2_ statements 😢.

>PS. Maybe BT firmware will get such functionality somewhen in the future. Seems like CYW43439 is an updated CYW4343W and firmware binaries are smaller, so there might be some patch RAM left (or something like that) to do so.

Do you know if there is somewhere a _roadmap_ of what is planned?

peterharperuk, 2025-02-25:
It's hard to get any changes made to the firmware.

matsobdev, 2025-02-27:
> Do you know if there is somewhere a roadmap of what is planned?

That was just my wild guess

Slion, 2025-03-08:
@bittailor Good catch!
@peterharperuk It does not hurt to ask 🫣

matsobdev, 2025-07-26:
But on the other hand, according to: https://datasheets.raspberrypi.com/pico/pico-2-product-brief.pdf, Pico 2 series has still around 15 years of in-production time.

mringwal, 2026-03-05:
@nathankellenicki Same answer as before, the CYW43439 on Pico W and Pico 2W cannot support anything modern. Right now I would suggest to use an nRF54L15 - devkits are easily available and there is also a tiny miniboard by Seeedstudio (but it lacks an on-board programmer unfortunately)