raspberrypi/pico-sdk #2316: WiFi disconnects and does not recover

mincequi, 2025-02-24:
Hi there,

i use the following that call to connect to my wifi which works flawlessly:
`cyw43_arch_wifi_connect_timeout_ms`

At some point in time (and at irregular intervals), i cannot access my pico anymore. Also MDNS responder does not react anymore, so i suspect that the pico-sdk won't reconnect, after its wifi got lost. Is that correct?

For the async variants of those connect functions, i can find this in the doc:
`You should subsequently check the status by calling \ref cyw43_wifi_link_status.`
I only can find cyw43_tcpip_link_status, so i assume that is misspelled?
Shall one subsequently poll the link_status and reconnect in case it's lost?

Thanks in advance.

peterharperuk, 2025-02-24:
It would be useful to get some logging off the device to confirm what it's doing. It could be stuck, crashed or something else. It could be that it's got a CYW43_EV_DISASSOC event for some reason. So it's definitely worth monitoring if cyw43_wifi_link_status changes. As a minimum you could turn the led on and off if it thinks the wifi status is active

mincequi, 2025-02-26:
@peterharperuk thanks for your hint. I added some "blinking codes" and could reproduce that behaviour with serial port attached and i received:
```
*** PANIC ***

sys_timeout: timeout != NULL, pool MEMP_SYS_TIMEOUT is empty
```

I fixed this with:
`#define MEMP_NUM_SYS_TIMEOUT (LWIP_NUM_SYS_TIMEOUT_INTERNAL + 4)`

But coming back to my original question: shall i be concerned about wifi getting disconnected? Or will it try to reconnect at any time, when it was connected once?

peterharperuk, 2025-02-26:
In theory you should be able to power the router off and on and the device will still be "connected" when it's available again. If the device see the CYW43_EV_DISASSOC event it won't automatically reconnect, you'd have to handle that in software.

Of course - there could be a real bug like this which is fixed in the latest release https://github.com/raspberrypi/pico-sdk/issues/2153

peterharperuk, 2025-06-02:
Not much more I can do with this without tracing. I suggest testing the latest version of develop

closed 2025-06-02 by peterharperuk