raspberrypi/pico-sdk #2317: RP2040 C++ project, with std::vectors, and run in RAM results in "hidden symbol `__dso_handle' isn't defined"

comfile-mike, 2025-02-27:
Project configuration:

[image]

Code:
```c++
#include <stdio.h>
#include "pico/stdlib.h"
#include <vector>

class Test
{
    std::vector<int> vector_of_ints;
};

static std::vector<Test> tests =
{
    Test(),
    Test()
};

int main()
{
    stdio_init_all();

    while (true) {
        printf("Hello, world!\n");
        sleep_ms(1000);
    }
}
```

Commenting out `pico_set_binary_type(vector_test no_flash)` in CMakeLists.txt will resolve the error.

lurch, 2025-02-27:
Duplicate of #1368 ?

comfile-mike, 2025-02-27:
Yeah, this appears to be a duplicate that issue.  Closing.

closed 2025-02-27 by comfile-mike