raspberrypi/pico-sdk #2320: ta_set_timeout may not work

yizi15, 2025-02-28:
SDK：2.1.0
chip：RP2350B

code:
'''
struct TimerHandle 
{
    repeating_timer_callback_t callback;
    alarm_pool_timer_t* timer;
    uint timer_alarm_num;
    int64_t period;
    uint64_t early_target;
};

extern struct TimerHandle timer_handle[3];

void alarm_irq_handler_0();

static void timer_start_direct(int _i)
{
    assert(timer_handle[_i].timer == NULL);
    timer_handle[_i].timer = ta_timer_instance(1);
    timer_handle[_i].timer_alarm_num = ta_hardware_alarm_claim_unused(timer_handle[_i].timer, true);
    alarm_pool_timer_t* timer = timer_handle[_i].timer; 
    uint timer_alarm_num =  timer_handle[_i].timer_alarm_num; 
    if(_i == 0)
    {
        ta_enable_irq_handler(timer, timer_alarm_num, alarm_irq_handler_0);
    }
    ta_clear_force_irq(timer, timer_alarm_num);
    timer_handle[_i].early_target = ta_time_us_64(timer) + 1100;
    ta_set_timeout(timer, timer_alarm_num, timer_handle[_i].early_target);
}

static  void timer_start(int _i, int64_t delay, repeating_timer_callback_t callback)
{
    timer_handle[_i].callback = callback;
    timer_handle[_i].period = delay;
     timer_start_direct(_i);
}
bool __time_critical_func(timer_callback)(__unused repeating_timer_t *rt) 
{
    if(duration_tick <= 1)
    {
        duration_tick = -1;
        return false;
    }
    else 
    {
        duration_tick --;
    }
    return true; // keep repeating
}

struct TimerHandle timer_handle[3];
void __time_critical_func(alarm_irq_handler_0)(void) 
{       
    #define _i 0                                           
    alarm_pool_timer_t* timer = timer_handle[_i].timer; 
    uint timer_alarm_num =  timer_handle[_i].timer_alarm_num; 
    int64_t period = timer_handle[_i].period; 
    ta_clear_force_irq(timer, timer_alarm_num);

    ta_clear_irq(timer, timer_alarm_num); 
    if(timer_callback(NULL)) 
    {                                   
        int64_t cur = (int64_t) ta_time_us_64(timer); 
        if( period > 0)
        {           
            timer_handle[_i].early_target = cur + period;
        }                
        else              
        {                 
            int64_t next = timer_handle[_i].early_target - period;        
            next = next > (cur + 3) ? next: cur + 5;    
            timer_handle[_i].early_target = next;      
        }     
        ta_set_timeout(timer, timer_alarm_num, timer_handle[_i].early_target);
              
    }                     
    else                    
    {                       
        ta_disable_irq_handler(timer, timer_alarm_num, alarm_irq_handler_0);      
        timer_handle[_i].timer = NULL;                                                
    } 
}
timer_start(0, 2500, NULL);
while (1)
  {
      printf("duration_tick %d\n", duration_tick);
      sleep_ms(1000);
  }
'''
and the log 
```
480000
479999
479999
479999
...
```
the sdk code
```
static inline void ta_set_timeout(alarm_pool_timer_t *timer, uint alarm_num, int64_t target) {
    // We never want to set the timeout to be later than our current one.
    uint32_t current = timer_time_us_32(timer_hw_from_timer(timer));
    uint32_t time_til_target = (uint32_t) target - current;
    uint32_t time_til_alarm = timer_hw_from_timer(timer)->alarm[alarm_num] - current;
    // Note: we are only dealing with the low 32 bits of the timer values,
    // so there is some opportunity to make wrap-around errors.
    //
    // 1. If we just passed the alarm time, then time_til_alarm will be high, meaning we will
    //    likely do the update, but this is OK since the alarm will have just fired
    // 2. If we just passed the target time, then time_til_target will be high, meaning we will
    //    likely not do the update, but this is OK since the caller who has the full 64 bits
    //    must check if the target time has passed when we return anyway to avoid races.
    if (time_til_target < time_til_alarm) {
        timer_hw_from_timer(timer)->alarm[alarm_num] = (uint32_t) target;
    }
}
```
When the current  == timer_hw_from_timer(timer)->alarm[alarm_num] , the function will not work

peterharperuk, 2025-02-28:
Please test with the latest sdk

peterharperuk, 2025-02-28:
Your code didn't compile and was incomplete which was a bit unhelpful. I had a quick go at running it and it didn't seem to reproduce the problem on master. Assumed to be a duplicate of #2186 or #2118

closed 2025-02-28 by peterharperuk