raspberrypi/pico-sdk #2322: Move btstack flash bank storage from the end of flash?

peterharperuk, 2025-02-28:
Saw this mentioned on the forum somewhere. By default the bt stack flash storage uses the last two sectors in flash. However picotool has started using the last page to workaround RP2350-E10.

We could.
- ignore the problem
- change the default to 3 sectors from the end
- move it somewhere else, e.g. end of the binary or inside the binary (I seem to remember circuit python do this?)

peterharperuk, 2025-02-28:
@will-v-pi Have I got my facts straight?

cc @kilograham Any opinion on what I should do?

kilograham, 2025-04-11:
merged into develop

kilograham, 2025-04-11:
@peterharperuk it occurs to me that this will break (lost credentials for) existing code that already has stored credentials - that said, currently downloading a new UF2 would also do the same, so maybe this isn't a big deal?

PR #2412 (2025-04-11): Change BT flash storage default for rp2350 A2
The workaround for errata RP2350-E10 overwrites the last block in flash. This will overwrite the BT flash storage causing a paired BT connection to fail. Move the default flash storage location to 3 sectors from the end of flash for RP2350 where A2 support is required.

This will require existing BT pairings to a Pico device to be removed and readded.

Fixes #2322
commit 6fea77f5 (2025-04-11): Change BT flash storage default for rp2350 A2 (#2412)

The workaround for errata RP2350-E10 overwrites the last block in flash.
This will overwrite the BT flash storage causing a paired BT connection
to fail. Move the default flash storage location to 3 sectors from the
end of flash for RP2350 where A2 support is required.

This will require existing BT pairings to a Pico device to be removed
and readded.

Fixes #2322