raspberrypi/pico-sdk #2325: Spin Lock Build Errors with Pico SDK on Raspberry Pi Pico 2 (RP2350)

crcz25, 2025-03-01:
### Issue Description
I'm encountering inconsistent build errors when using the Pico SDK with the new Raspberry Pi Pico 2 (RP2350). The initial build attempt fails with multiple errors related to missing spin lock implementation for the RP2350 platform. However, in a second build attempt, the process succeeds without changes, which is unexpected and confusing. What could be causing these intermittent build failures? Are there known issues with spin locks on RP2350, or what might be happening?

### Error details
When building with PICO_BOARD=pico2 and PICO_PLATFORM=rp2350, I get the error below. This error occurs across multiple steps (see the output from the terminal, [terminal_log.txt](https://github.com/user-attachments/files/19040208/terminal_log.txt)). This is also my building script ([build.txt](https://github.com/user-attachments/files/19040215/build.txt)), and my CMakeLists ([CMakeLists.txt](https://github.com/user-attachments/files/19040217/CMakeLists.txt)).

`#error no SW_SPIN_TRY_LOCK available for PICO_USE_SW_SPIN_LOCK on this platform`

### Environment

- Board: Raspberry Pi Pico 2 (RP2350)
- OS: Ubuntu 24.04 (Docker container)
- Using Pico SDK from the official repository
- Build: CMake and arm-none-eabi-gcc (version 13.2.1) using build.sh
- Platform: rp2350-arm-s
- Project: Micro-ROS implementation (currently only loading the library and example code from the Micro-ROS tutorial). The publisher/subscriber works when I'm able to compile, so no issues there.

### Reproduction steps
Configure CMake with PICO_BOARD=pico2 and PICO_PLATFORM=rp2350
Run build
Build fails at compilation stage with the spin lock errors

Observations
- The first build attempt fails due to missing spin lock implementation for RP2350.
- The second attempt succeeds without any changes.
- No modifications were made between attempts, making the behavior inconsistent.

Any insights or debugging suggestions would be greatly appreciated!

lurch, 2025-03-03:
Not _directly_ related to your original question, but I see that https://micro.ros.org/docs/overview/hardware/ only lists RP2040 and doesn't list RP2350 yet.

crcz25, 2025-03-05:
Thanks @lurch, for pointing out the other issue, I can now compile the code successfully!

Previously, when compiling for the RP2040, I had set CFLAGS and CXXFLAGS as follows:

```
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -ffunction-sections -fdata-sections")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -ffunction-sections -fdata-sections")
```
After commenting out these lines, the compilation worked.

To be honest, I don’t fully understand this part of CMake, and I don’t even remember why I added these flags in the first place. As far as I know, they help reduce the final binary size, am I correct? 

Also, I understood that issue #452 from pico-feedback was related to the distro (ClearLinux), right? For reference, I’m compiling my code inside Docker using Ubuntu 24.04.2 LTS, while my host OS is Pop!_OS 22.04 LTS. 

I'm not sure how to proceed with this issue, I think it can be closed, but not sure if any further action needs to be taken. Could these flags just be incompatible with the new Pico? 🤔

keithr0, 2025-03-15:
I have a similar problem, compilation fails every time with the following error:
`no SW_SPIN_TRY_LOCK available for PICO_USE_SW_SPIN_LOCK on this platform`
at lines 152, 193, and 213 in pico-sdk/src/rp2_common/hardware_sync_spin_lock/include/hardware/sync/spin_lock.h.
From the code is would appear that `__ARM_ARCH_8M_MAIN__` is not defined.
PICO_PLATFORM = rp2350-arm-s, PICO_BOARD = seeed_xiao_rp2350. System Ubuntu under 5.15.167.4-microsoft-standard-WSL2.
Project is a GPS application with CPU1 communicating with the GPS CPU0 calculating and displaying  the output. The line  causing the error is `#include "pico/multicore.h"    //`

lurch, 2025-03-17:
@keithr0 Have you tried deleting your `build` directory and re-running `cmake`?

> System Ubuntu under 5.15.167.4-microsoft-standard-WSL2.

I don't think we officially support WSL2. Are you able to reproduce the same problem using https://github.com/raspberrypi/pico-vscode ?

kilograham, 2025-03-18:
Hmm, i wonder if this is a ninja rhing.... can you try with a clean build directory, and

passing `-G "Unix Makefiles"` to CMake too; and also post the full `cmake` and `make` build output of your original if that doesn't fix it.

kilograham, 2025-03-18:
also `cmake --version` and `cmake -G` output would be helpful

lurch, 2025-04-29:
@crcz25 So if you've fixed the issue, then I guess that means that you're no longer able to reproduce it; and so this issue can be closed?

crcz25, 2025-05-01:
Yep, I think it can be closed. Thanks!!

closed 2025-05-01 by lurch

rtarun1, 2025-10-01:
> Yep, I think it can be closed. Thanks!!

@crcz25 Were u able to get micro ros working with RP2350?