raspberrypi/pico-sdk #2330: Linux RP2350 Builds Fail

jadczak, 2025-03-03:
The default example blink program fails to build on linux.  It fails on SDK 2.1 and 2.1.1

Trying to run cmake manually suggest that some unrecognized options are being passed to the compiler (`-m64` and `-mrelax-cmpxchg-loop`)

[image]

[image]

lurch, 2025-03-04:
Which Linux distro are you using? What hardware platform are you running on?
Which setup instructions did you follow to get to your current situation?

jadczak, 2025-03-04:
1. Clear Linux
2. Trying compile for my pico 2.
3. followed the basic getting started.  All the dependencies were installed so I didn't need to add anything to my system. https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

This is (well was) a clean install of both VS Code.  I gave VS Codium a shot too for kicks.  Both yield the same result after creating a new project from the example.

Compilation works perfectly fine on my windows machine.

jadczak, 2025-03-04:
after poking around I *think* the issue stems from Clear Linux.

I rebooted my system and ran `echo $CFLAGS` from a fresh terminal and got the following:

```
-g -O3 -feliminate-unused-debug-types -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -Wformat -Wformat-security -m64 -fasynchronous-unwind-tables -Wp,-D_REENTRANT -ftree-loop-distribute-patterns -Wl,-z,now -Wl,-z,relro -fno-semantic-interposition -ffat-lto-objects -fno-trapping-math -Wl,-sort-common -Wl,--enable-new-dtags -mrelax-cmpxchg-loop -ffunction-sections
```

So, it seems when the new terminal process is launched during the pico compilation, all those flags come along for the ride.

I'm not a CMAKE jockey so I don't know how to tell CMAKE to clear out the CFLAGS.

jadczak, 2025-03-04:
And some more throwing commands at the wall to see what sticks...

 `cmake -DCMAKE_C_FLAGS="" -DCMAKE_CXX_FLAGS=""  ..` 

yields the following:

```
PICO_SDK_PATH is /home/alex/.pico-sdk/sdk/2.1.1
Target board (PICO_BOARD) is 'pico2'.
Using board configuration from /home/alex/.pico-sdk/sdk/2.1.1/src/boards/include/boards/pico2.h
Pico Platform (PICO_PLATFORM) is 'rp2350-arm-s'.
-- The C compiler identification is GNU 14.2.1
-- The CXX compiler identification is GNU 14.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/alex/.pico-sdk/toolchain/14_2_Rel1/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/alex/.pico-sdk/toolchain/14_2_Rel1/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/alex/.pico-sdk/toolchain/14_2_Rel1/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
-- Found Python3: /bin/python (found version "3.13.2") found components: Interpreter
TinyUSB available at /home/alex/.pico-sdk/sdk/2.1.1/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/alex/.pico-sdk/sdk/2.1.1/lib/btstack
cyw43-driver available at /home/alex/.pico-sdk/sdk/2.1.1/lib/cyw43-driver
lwIP available at /home/alex/.pico-sdk/sdk/2.1.1/lib/lwip
mbedtls available at /home/alex/.pico-sdk/sdk/2.1.1/lib/mbedtls
-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/alex/source/rp-pico/blink/build
```

But the build fails.  Truncated log below.

```
Executing task: /home/alex/.pico-sdk/ninja/v1.12.1/ninja -C /home/alex/source/rp-pico/blink/build 

ninja: Entering directory `/home/alex/source/rp-pico/blink/build'
[7/62] Building C object CMakeFiles/blink.dir/home/alex/.pico-sdk/sdk/2.1.1/src/common/hardware_claim/claim.c.o
FAILED: CMakeFiles/blink.dir/home/alex/.pico-sdk/sdk/2.1.1/src/common/hardware_claim/claim.c.o 
/home/alex/.pico-sdk/toolchain/14_2_Rel1/bin/arm-none-eabi-gcc -DLIB_BOOT_STAGE2_HEADERS=1 -DLIB_PICO_ATOMIC=1 -DLIB_PICO_BIT_OPS=1 -DLIB_PICO_BIT_OPS_PICO=1 -DLIB_PICO_CLIB_INTERFACE=1 -DLIB_PICO_CRT0=1 -DLIB_PICO_CXX_OPTIONS=1 -DLIB_PICO_DIVIDER=1 -DLIB_PICO_DIVIDER_COMPILER=1 -DLIB_PICO_DOUBLE=1 -DLIB_PICO_DOUBLE_PICO=1 -DLIB_PICO_FLASH=1 -DLIB_PICO_FLOAT=1 -DLIB_PICO_FLOAT_PICO=1 -DLIB_PICO_FLOAT_PICO_VFP=1 -DLIB_PICO_INT64_OPS=1 -DLIB_PICO_INT64_OPS_COMPILER=1 -DLIB_PICO_MALLOC=1 -DLIB_PICO_MEM_OPS=1 -DLIB_PICO_MEM_OPS_COMPILER=1 -DLIB_PICO_NEWLIB_INTERFACE=1 -DLIB_PICO_PLATFORM=1 -DLIB_PICO_PLATFORM_COMPILER=1 -DLIB_PICO_PLATFORM_PANIC=1 -DLIB_PICO_PLATFORM_SECTIONS=1 -DLIB_PICO_PRINTF=1 -DLIB_PICO_PRINTF_PICO=1 -DLIB_PICO_RUNTIME=1 -DLIB_PICO_RUNTIME_INIT=1 -DLIB_PICO_STANDARD_BINARY_INFO=1 -DLIB_PICO_STANDARD_LINK=1 -DLIB_PICO_STDIO=1 -DLIB_PICO_STDIO_UART=1 -DLIB_PICO_STDLIB=1 -DLIB_PICO_SYNC=1 -DLIB_PICO_SYNC_CRITICAL_SECTION=1 -DLIB_PICO_SYNC_MUTEX=1 -DLIB_PICO_SYNC_SEM=1 -DLIB_PICO_TIME=1 -DLIB_PICO_TIME_ADAPTER=1 -DLIB_PICO_UTIL=1 -DPICO_32BIT=1 -DPICO_BOARD=\"pico2\" -DPICO_BUILD=1 -DPICO_CMAKE_BUILD_TYPE=\"Release\" -DPICO_COPY_TO_RAM=0 -DPICO_CXX_ENABLE_EXCEPTIONS=0 -DPICO_NO_FLASH=0 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_RP2350=1 -DPICO_TARGET_NAME=\"blink\" -DPICO_USE_BLOCKED_RAM=0 -I/home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/pico_atomic/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/common/pico_stdlib_headers/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_gpio/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/common/pico_base_headers/include -isystem /home/alex/source/rp-pico/blink/build/generated/pico_base -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/boards/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2350/pico_platform/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2350/hardware_regs/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_base/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/pico_platform_compiler/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/pico_platform_panic/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/pico_platform_sections/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_dcp/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2350/hardware_structs/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_rcp/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/common/hardware_claim/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_sync/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_sync_spin_lock/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_irq/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_uart/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_resets/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_clocks/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_pll/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_vreg/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_watchdog/include -isystem /home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_ticks/include -isystem /home/ [... cut ...]
/home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_sync_spin_lock/include/hardware/sync/spin_lock.h: In function 'spin_unlock_unsafe':
/home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_sync_spin_lock/include/hardware/sync/spin_lock.h:286:5: error: implicit declaration of function 'SW_SPIN_LOCK_UNLOCK'; did you mean 'SW_SPIN_LOCK_NUM'? [-Wimplicit-function-declaration]
  286 |     SW_SPIN_LOCK_UNLOCK(lock);
      |     ^~~~~~~~~~~~~~~~~~~
      |     SW_SPIN_LOCK_NUM
[8/62] Building C object CMakeFiles/blink.dir/home/alex/.pico-sdk/sdk/2.1.1/src/rp2_common/hardware_sync/sync.c.o
```

lurch, 2025-03-04:
> `#error no SW_SPIN_TRY_LOCK available for PICO_USE_SW_SPIN_LOCK on this platform`

In #2325 Graham suggests that this error might be caused by stale artefacts in your build directory, i.e. you need to delete your build directory and try again.

Moving this issue from pico-sdk to pico-feedback, as it's an environment issue rather than an SDK bug.