raspberrypi/pico-sdk #2331: Missing RASPBERRY_PI_PICO.h in src/boards/include/boards/ on master branch

Beebuzzer05, 2025-03-04:
[image]

Iam unable to build projects for the Raspberry Pi Pico because the RASPBERRY_PI_PICO.h header file is missing from the src/boards/include/boards/ directory in the master branch of the Pico SDK repository.

I have verified this by:

Cloning the latest version of the Pico SDK.
Checking the src/boards/include/boards/ directory locally.
Confirming the file's absence in the online repository: [https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards](https://www.google.com/url?sa=E&source=gmail&q=https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards) (see attached screenshot).
This is a critical issue as it prevents users from building projects for the standard Raspberry Pi Pico.

Please investigate and restore the missing header file as soon as possible.

Environment:

Operating System: (Please specify your OS, e.g., Ubuntu 22.04, macOS Ventura)
Git Version: (Please run git --version and provide the output)
Attachments:

Screenshot of the src/boards/include/boards/ directory in the online repository, showing the missing file.
Additional Notes:

If you have any other relevant information, such as any recent changes you made to your system or any error messages you encountered, please include them as well.
You can also mention that you have tried re-cloning the repository and checking for network issues, but the problem persists.
Remember to replace the placeholders for your operating system and Git version with the actual values.

By providing a clear and detailed report, you'll help the maintainers understand and resolve the issue quickly.

d-c-d, 2025-03-04:
I would guess you are looking for the file "pico.h" which is there

lurch, 2025-03-04:
> I would guess you are looking for the file "pico.h" which is there

Indeed. Please see the documentation around `PICO_BOARD` in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf and https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

closed 2025-03-04 by lurch