raspberrypi/pico-sdk #2332: ta_disable_irq_handler hardware_alarm_unclaim not use arg `timer`

yizi15, 2025-03-04:
version:master
chip: rp2350B

```
static inline void ta_disable_irq_handler(alarm_pool_timer_t *timer, uint alarm_num, irq_handler_t irq_handler) {
    uint irq_num = timer_hardware_alarm_get_irq_num(timer, alarm_num);
    timer_hw_from_timer(timer)->armed = 1u << alarm_num; // disarm the timer
    hw_clear_bits(&timer_hw_from_timer(timer)->inte, 1u << alarm_num);
    irq_set_enabled(irq_num, true);
    irq_remove_handler(irq_num, irq_handler);
    hardware_alarm_unclaim(alarm_num);
}
```
When the timer is timer1, the hardware_alarm_unclaim also unclaim the timer0

lurch, 2025-03-04:
Looks like using `timer_hardware_alarm_unclaim` https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_timer/timer.c#L30 instead of `hardware_alarm_unclaim` might fix the problem? 🤷

kilograham, 2025-04-03:
merged into develop

closed 2025-04-03 by kilograham