raspberrypi/pico-sdk #2333: SDK 2.1.1 RP2350, writing to flash changes QSPI pad settings

UKTailwind, 2025-03-06:
Writing to flash changes the QSPI pad settings back to default.
I've worked round this by making the following change to flash.c

`typedef struct flash_rp2350_qmi_save_state {
    uint32_t timing;
    uint32_t rcmd;
    uint32_t rfmt;
    uint32_t io[6];
} flash_rp2350_qmi_save_state_t;

static void __no_inline_not_in_flash_func(flash_rp2350_save_qmi_cs1)(flash_rp2350_qmi_save_state_t *state) {
    state->timing = qmi_hw->m[1].timing;
    state->rcmd = qmi_hw->m[1].rcmd;
    state->rfmt = qmi_hw->m[1].rfmt;
    state->io[0]=pads_qspi_hw->io[0];
    state->io[1]=pads_qspi_hw->io[1];
    state->io[2]=pads_qspi_hw->io[2];
    state->io[3]=pads_qspi_hw->io[3];
    state->io[4]=pads_qspi_hw->io[4];
    state->io[5]=pads_qspi_hw->io[5];
}

static void __no_inline_not_in_flash_func(flash_rp2350_restore_qmi_cs1)(const flash_rp2350_qmi_save_state_t *state) {
    if (flash_devinfo_get_cs_size(1) == FLASH_DEVINFO_SIZE_NONE) {
        // Case 1: The RP2350 ROM sets QMI to a clean (03h read) configuration
        // during flash_exit_xip(), even though when CS1 is not enabled via
        // FLASH_DEVINFO it does not issue an XIP exit sequence to CS1. In
        // this case, restore the original register config for CS1 as it is
        // still the correct config.
        qmi_hw->m[1].timing = state->timing;
        qmi_hw->m[1].rcmd = state->rcmd;
        qmi_hw->m[1].rfmt = state->rfmt;
        pads_qspi_hw->io[0]=state->io[0];
        pads_qspi_hw->io[1]=state->io[1];
        pads_qspi_hw->io[2]=state->io[2];
        pads_qspi_hw->io[3]=state->io[3];
        pads_qspi_hw->io[4]=state->io[4];
        pads_qspi_hw->io[5]=state->io[5];
        } else {
        // Case 2: If RAM is attached to CS1, and the ROM has issued an XIP
        // exit sequence to it, then the ROM re-initialisation of the QMI
        // registers has actually not gone far enough. The old XIP write mode
        // is no longer valid when the QSPI RAM is returned to a serial
        // command state. Restore the default 02h serial write command config.
        qmi_hw->m[1].wfmt = QMI_M1_WFMT_RESET;
        qmi_hw->m[1].wcmd = QMI_M1_WCMD_RESET;
    }
}
#endif
`

Wren6991, 2025-07-16:
This seems reasonable except that the restore of the pads should not depend on the CS1 availability.

It is adding a bit more code size and stack frame size, so I would prefer to make the XIP setup function (boot2) more configurable instead of adding save/restore everywhere. On the other hand the XIP setup function has usually already run when we get here, and if someone goes and modifies the pads afterwards then it seems reasonable to preserve those changes.

Wren6991, 2025-07-16:
The main drawback is this breaks code which was relying on these functions to initialise the pads, like RAM binaries which use flash (which is not a use case we handle well at the moment). We could expose something similar to `flash_enable_xip_via_boot2()` as a non-static symbol.

PR #2565 (2025-07-20): hardware_flash: preserve QSPI pad state over flash access calls.
Add new function, flash_start_xip(), which explicitly performs a first-time XIP setup (including initialising pads). This is mostly useful for PICO_NO_FLASH=1 binaries where there is actually an attached external flash.

Fixes #2333