raspberrypi/pico-sdk #2336: Enable partial code encryption for larger binaries

Slion, 2025-03-08:
The [encrypted bootloader example] demonstrate how to encrypt your binaries and have them securely decrypted in place from SRAM. However, as it is, it only works for the smallest binaries below 448KB minus your stack and heap budget, unless I'm mistaken.

Luckily, as hinted by @kilograham in that [forum thread], there could be ways to partially encrypt your binaries. Is there any chance that we could get tools support and an example to do just that? 

Ideally, one should be able to encrypt only the application code and not all the SDK and other libraries it is calling. Though my understanding of the linker, ELF and linker scripts is currently far too limited to understand if that could be doable at all.

[encrypted bootloader example]: https://github.com/raspberrypi/pico-examples/tree/master/bootloaders/encrypted
[forum thread]: https://forums.raspberrypi.com/viewtopic.php?t=377074

will-v-pi, 2025-04-16:
Note this would probably need a new SDK binary type, to have a signed XIP & SRAM binary with an entry point in SRAM, so it can decrypt data and lock OTP pages containing the key before any XIP is executed.

Slion, 2025-08-16:
Could possibly be achieved by doing something like:
```
__attribute__((section(".encrypt")))
void secret_function() {
        // Proprietary logic
}
```

Extract encrypt section to actually encrypt it:

```
arm-none-eabi-objcopy -O binary --only-section=.encrypt firmware.elf encrypted.bin
```

Linker script should do something like: 

```
.encrypt (NOLOAD) : {
    KEEP(*(.encrypt))
} > NONE
```