raspberrypi/pico-sdk #2337: amethyst_fpga.h // FIXME delete this file before release

d-c-d, 2025-03-08:
`src/boards/include/boards/amethyst_fpga.h` has a comment:

`// FIXME delete this file before release (board file for Amethyst FPGA platform)`

this file is part of multiple repositories (e.g. circuitpython) so it may be hard to delete it now :-)

Do we want to re-write history and delte the comment, or just delete the file ?

if the board file is deleted - there are a few lines in 

`src/rp2_common/pico_crt0/crt0_riscv.S` and
`src/rp2_common/pico_bootrom/include/pico/bootrom.h`
   that could be deleted

lurch, 2025-03-09:
> if the board file is deleted - there are a few lines in
> 
> `src/rp2_common/pico_crt0/crt0_riscv.S` and `src/rp2_common/pico_bootrom/include/pico/bootrom.h` that could be deleted

Also in `tools/check_board_header.py` 😉

kilograham, 2025-03-09:
the comment should be deleted - we decided to keep the file

kilograham, 2025-04-04:
merged into develop

PR #2341 (2025-03-18): Remove FIXME delete comment
Fixes #2337
per @kilograham
commit 6a564808 (2025-03-18): Remove FIXME delete comment (#2341)

Fixes #2337
per @kilograham
closed 2025-04-04 by kilograham