raspberrypi/pico-sdk #2338: irq clear prev not working

pauldevine, 2025-03-09:
On the rp2350b I'm synchronizing two state machines on different PIO instances so that they can address different pin ranges. 0-31 with one state machine and 32-47 on the other. I'm able to successfully set an interrupt and have it triggered on the second machine across the PIO boundary. But I'm not able to clear it with `irq clear`. The compiler won't allow me to reference the previous pio block with:

```
.wrap_target
    wait 0 irq prev 0      ; Wait for IRQ0
    irq clear prev 0        ; Clear IRQ0
.wrap
```

that results in a syntax error `15-18: syntax error, unexpected prev, expecting ( or identifier or integer.` I get the same syntax error if I try the documented `irq clear 0 prev`. Since the word prev seems to work for the other statements when it precedes the interrupt number I assumed that syntax should be valid here.

I can get the pio assembly to compile if I use `irq prev clear 0` however that doesn't clear the interrupt. 

If instead I clear the interrupt immediately with `wait 1 irq prev 0` that works as expected. I was hoping to have a few pio commands executed before clearing the interrupt as I'm trying to get a bus to the right state before continuing the other state machine.

lurch, 2025-03-10:
Similar issue to https://github.com/raspberrypi/pico-feedback/issues/440 ?

pauldevine, 2025-03-10:
That earlier comment is pointing out a documentation error. I believe I'm experiencing a functional error.

KewlKris, 2025-03-11:
I also had trouble with IRQs between state machines in different blocks (even when synchronizing with CTRL.NEXTPREV_CLKDIV_RESTART).

I eventually used `wait JMPPIN + n` (to wait for a signal) and `jmp PIN` (to branch on a signal) to synchronize and communicate between state machines. I wish I could get things working with an internal solution rather than using external pins, though.

kilograham, 2025-07-08:
Can you please attach a small example showing the problem; does it work as expected without the `next`?

pauldevine, 2025-07-09:
I can put together a small reproducer over the weekend.

pauldevine, 2025-07-13:
[irq_test.zip](https://github.com/user-attachments/files/21206933/irq_test.zip)
Attached is a small reproducer. If you run the existing setup as shown you'll get output like:
Starting IRQ test
IRQ 0 state: 1
IRQ 0 state: 1
IRQ 0 state: 1
IRQ 0 state: 1
IRQ 0 state: 1
IRQ 0 state: 1
IRQ 0 state: 1
IRQ 0 state: 1
IRQ 0 state: 1
IRQ 0 state: 1

which shows the IRQ never being cleared. If in the consumer.pio file you change the line `wait 0 irq prev 0      ; Wait for IRQ0` to instead be `wait 1 irq prev 0      ; Wait for IRQ0` you instead get the output:
Starting IRQ test
IRQ 0 state: 0
IRQ 0 state: 0
IRQ 0 state: 0
IRQ 0 state: 0
IRQ 0 state: 0
IRQ 0 state: 0
IRQ 0 state: 0
IRQ 0 state: 0
IRQ 0 state: 0

Given my understanding of the pio commands I would the first example wait to be clearing the IRQ the same as the second wait statement..

kilograham, 2025-07-20:
You didn't answer:

> does it work as expected without the next?

which would have been a good data point. The documentation could perhaps be clearer, but for

`wait <pol> irq <n>`  

`<pol>` is still the polarity; the IRQ is just cleared if you wait with a polarity of 1. i.e. `wait 0 irq <n>` waits for a 0

pauldevine, 2025-07-21:
Sorry for not being clear, yes it does work without the next, but that requires both .pio functions to be on the same pio block, which wasn't possible in my case as the rp2350 pin ranges differed. 

Your comment:
> <pol> is still the polarity; the IRQ is just cleared if you wait with a polarity of 1. i.e. wait 0 irq <n> waits for a 0

That doesn't match the observed behavior on the same pio block nor the documentation. In https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf on page 888:

> WAIT x IRQ behaves slightly differently from other WAIT sources:
> • If Polarity is 1, the selected IRQ flag is cleared by the state machine upon the wait condition being met.

This is what I have observed when on the same PIO instance. If you fire the interrupt a `wait 0 irq 1` it allows you to issue additional PIO instructions and then `irq clear 1` after you've taken actions that needed to be issued prior to the other state machine proceeding. Without this functionality you end up creating race conditions between the different state machines. It's this behavior that didn't work with the prev/next but did work with the statements on the same PIO instance.

kilograham, 2025-07-21:
I had tried it earlier, and with `wait 0 irq 0` on a single PIO i got the same behavior as with multiple PIO blocks. Ditto with `wait 1 irq 0`

closed 2026-01-05 by kilograham