raspberrypi/pico-sdk #2355: Inconsistent use of PICO_RP2040 macro in embedded_start_block.inc.S

eightycc, 2025-03-18:
In `src/rp2_common/pico_crt0/embedded_start_block_inc.S` the macro `PICO_RP2040` is tested with `#ifdef`. Elsewhere in the SDK, `PICO_RP2040` is expected to have a value of 0 for RP2350 and 1 for RP2040. `#ifdef` usage results in an incorrect `ARM.attributes` section generated when compiling `crt0.S` with `PICO_RP2040=0` and `PICO_RP2350=1` resulting in an un-bootable UF2.

kilograham, 2025-03-18:
merged into develop

lurch, 2025-03-18:
@kilograham On a similar note, by doing some `git grep` I see that most places in the SDK use `#if PICO_RP2350` but https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_rcp/include/hardware/rcp.h#L19 uses `#if defined(PICO_RP2350)` - does that also need fixing up?

kilograham, 2025-03-18:
> [@kilograham](https://github.com/kilograham) On a similar note, by doing some `git grep` I see that most places in the SDK use `#if PICO_RP2350` but [...]

Actually that one should just be `#if HAS_REDUNDANCY_COPROCESSOR` as we now have feature flags like that (which also takes care of the RISC-V ness.

Probably worth looking at any remaining `PICO_RP2350` OR `PIO_RP2040` tests

kilograham, 2025-03-18:
re-opening for me to go and have a quick look at other places

kilograham, 2025-07-21:
none others found

PR #2356 (2025-03-18): Fix usage of PICO_RP2040 macro.
Fixes #2355.

Corrects usage of PICO_RP2040 macro in `embedded_start_block.inc.S` by testing value instead of testing for definition.
closed 2025-07-21 by kilograham