raspberrypi/pico-sdk #2365: Unstable to connect 2.4Ghz network AP via micropython firmware

yoyojacky, 2025-03-20:
On my deivce : Pico 2 w 
Micropython version :MicroPython v1.25.0-preview.389.gf187c77da on 2025-03-14; Raspberry Pi Pico 2 W with RP2350 

part of demo code: 
[image]
result:
[image]

sometimes it works , but most of time it is out of work...

lurch, 2025-03-20:
I believe questions about MicroPython are best asked at https://github.com/orgs/micropython/discussions

yoyojacky, 2025-03-21:
> I believe questions about MicroPython are best asked at https://github.com/orgs/micropython/discussions

OK, thanks for your tips .

yoyojacky, 2025-03-21:
> Can you try running some c example code and see if you have a problem with that? e.g. https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/iperf

Ok, I will try it, I just shift to C enviroment right now. 

[image]
but I met this error due to I am not expert in C .. but still thanks for your help.

closed 2025-03-21 by yoyojacky

peterharperuk, 2025-03-21:
You have the value of PICO_BOARD wrong. It should be pico2_w (with the underscore)