raspberrypi/pico-sdk #2366: pico_btstack_make_gatt_header broken when specifying files without path

Slion, 2025-03-20:
While working on [pico-demo-gatt-service] I realized that `pico_btstack_make_gatt_header` is broken unless you specify a path for your GATT file. File names without path break the build with the following error:

```
Keyword WORKING_DIRECTORY requires a value, but none was given.
```

To reproduce the issue you can use the example linked above and remove the path `${CMAKE_CURRENT_LIST_DIR}/` from the GATT file name.

[pico-demo-gatt-service]: https://github.com/Slion/pico-demo-gatt-service

kilograham, 2025-03-29:
I think it is reasonable to check in this case, since the error message is confusing, and we're explicitly splitting out the path in our code and expecting it to be present (this is subtly different from just the filename not being found)

kilograham, 2025-04-04:
merged into develop

PR #2385 (2025-04-04): Fix GATT_PATH error #2366
Set default GATT path to CMAKE_CURRENT_LIST_DIR if not specified

This sets the GATT_PATH variable to the calling directory when just a filename is provided with no path, as CMAKE_CURRENT_LIST_DIR refers to the calling CMakeLists.txt, not the function CMakeLists.txt

Fixes #2366
commit 701c1933 (2025-04-04): Set default GATT path to CMAKE_CURRENT_LIST_DIR if not specified (#2385)

This ensures the GATT_PATH variable is set to the calling directory when just a filename is provided with no path

Fixes #2366
closed 2025-04-04 by kilograham