raspberrypi/pico-sdk #2372: PSRAM: SDK support for bringing up PSRAM

Gadgetoid, 2025-03-25:
Between us (Pimoroni), SparkFun and Adafruit we've been pushing for PSRAM support in MicroPython (or CircuitPython in Adafruit's case but potato pota..to), since the extra RAM benefits are huge.

See: https://github.com/micropython/micropython/pull/15620

However as part of this there's some bringup that (as mentioned also by MicroPython's maintainers) feels more like Pico SDK's responsibility.

This includes detecting and getting the size of a PSRAM chip:

https://github.com/micropython/micropython/blob/293d17e4463142588ce4fd45f82b0bd628735676/ports/rp2/rp2_psram.c#L9-L78

And setting up PSRAM for general use (it then gets used to store additional heap for MicroPython):

https://github.com/micropython/micropython/blob/293d17e4463142588ce4fd45f82b0bd628735676/ports/rp2/rp2_psram.c#L80-L167

This is all rather spicy code that could benefit from an official canonical solution for bringup and the additional test coverage and exercising that SDK inclusion would imply.

Tighter integration into the SDK could also do things like assert when attempting to write to flash from a PSRAM resident address (bad times) or even attempt to handle cases like this transparently.

All three of us (to the best of my knowledge) have shipping products using this (or similar) PSRAM code in either MicroPython or CircuitPython and seem to have ironed out any quirks.

Are there any plans for this? If not, could it be considered? Thank you!

kilograham, 2025-03-25:
Seems reasonable, putting in 2.2.0 for now

sfe-SparkFro, 2025-03-25:
Also worth adding, the Arduino Pico port also has its own PSRAM implementation: https://github.com/earlephilhower/arduino-pico/blob/ecf2b2e39f04d343e13be9151b67232e6e3643a8/cores/rp2040/psram.cpp with a test example here: https://github.com/earlephilhower/arduino-pico/blob/ecf2b2e39f04d343e13be9151b67232e6e3643a8/libraries/rp2350/examples/PSRAMTest/PSRAMTest.ino

Totally agree on having an implementation in the Pico SDK that everyone else can leverage. We at SparkFun have put PSRAM on basically every RP2350 board we're selling.