raspberrypi/pico-sdk #2373: unique_id is empty if called by C++ static initializers

byteit101, 2025-03-27:
I had been stumped by the fact that this code was constantly returning 0000000000000000:

```cxx
#include "pico/unique_id.h"
struct unique_board_id_hex_str
{
	char c_str[PICO_UNIQUE_BOARD_ID_SIZE_BYTES * 2 + 1];
	unique_board_id_hex_str()
	{
		pico_get_unique_board_id_string(c_str, sizeof(c_str));
	}
};
const unique_board_id_hex_str usbd_serial_str{};

// later, in main:  usbd_serial_str.c_str == "0000000000000000"
```

After much confusion, I realized that `pico_get_unique_board_id_string` doesn't initialize `retrieved_id` and `_retrieve_unique_id_on_boot` is instead set up as a constructor:

```c
/// unique_id.c
static pico_unique_board_id_t retrieved_id;
static void __attribute__((constructor)) _retrieve_unique_id_on_boot(void) { retrieved_id = /* snip */; }
void pico_get_unique_board_id_string(char *id_out, uint len) { /* snip*/ }
```

It appears that `pico_get_unique_board_id_string` either must check for an uninitialized value and initialize it itself, or that the constructor must be run earlier via numbering it:

Example:
```c
static void __attribute__((constructor(1000))) _retrieve_unique_id_on_boot(void) { retrieved_id = /* snip */; }
```

This will let users order the sequence of startup (example follows), and will before un-numbered constructors/static c++ initialization by default. I can create a PR for either approach, but I wasn't sure which option was preferred by the maintainers.

```cxx
[[gnu::init_priority(2000)]]
const unique_board_id_hex_str usbd_serial_str{};
```

lurch, 2025-03-27:
> It appears that `pico_get_unique_board_id_string` either must check for an uninitialized value and initialize it itself

I believe that wouldn't work on RP2040, as in order to read the flash-chip's unique ID you need to turn off XIP mode, and that would probably cause your code to crash if you're currently executing from flash. Which is exactly why it does the "injects a call to .... to run before main" https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#detailed-description-23

So I guess you'd need to use the other approach you describe.

kilograham, 2025-03-29:
merged into develop

PR #2379 (2025-03-29): Move unique_id (configurably) earlier in the static init process
Fixes #2373

Move unique_id (configurably) earlier in the static initialization process . This enables using the unique_id in C++ static initializers by default, and avoid the static initialization order fiasco.

I added the configuration in the off chance that someone would need to adjust it, but I think 1000 is probably a good-enough default. Values range from 101-65535, or "none", which is after 65535. Normal C++ static initializers are "none". Users can still manually specify to be earlier (101-999) without any configuration  [... cut]
closed 2025-03-29 by kilograham